aluminum-ice/pwnagotchi #4: [BUG] No pwnagotchi peering

waterfallm, 2023-03-21:
**Describe the bug**
No pwnagotchi peering since upgrading to 1.6..2

**To Reproduce**
Steps to reproduce the behavior:
1. start 2nd pownagotchi device up

**Expected behavior**
Peer information displayed

**Environment (please complete the following information):**
 - 1.6.2 x 2
 - PI Zero W 2 off

**Additional context**
Had previously seen peering information running 1.5.5 tried with and without grid enabled.

Note: I also suspect the Wifi card has different setting under this version as the Wifi seems to find less devices.

waterfallm, 2023-03-28:
Finally got around to testing again with 1.5.5 installed and peering returns

[image]

RasTacsko, 2023-03-29:
I have the same issue.
1.5.1 and 1.5.5 can see each other but cannot see the 1.6.2 version.
1.6.2 version can see both older ones.
So the peer detection is working, but it seems the advertisement part is not.

RasTacsko, 2023-04-04:
Yepp, I used the same config.toml with 1.5.x release and after fresh install the 1.6.2 version almost immediately recognised

waterfallm, 2023-04-06:
> So the peer detection is working, but it seems the advertisement part is not.

Great i hadn't tried this way around yet i'll introduce a 1.6.2 version into the mix and see what i can see.

waterfallm, 2023-04-08:
> Great i hadn't tried this way around yet i'll introduce a 1.6.2 version into the mix and see what i can see.

tried with 1.7.0 and 1.6.2

1.7.0 will display in peers tab and also on home screen

1.6.2 doesn't see 1.7.0 peer in either homer or peers tab.

rsun0525, 2023-08-22:
After implementing 1.7.5, the peers are working perfectly for me now, I think you can close this issue.

waterfallm, 2023-09-13:
Only just seen this.. tested and working fine..