aluminum-ice/pwnagotchi #5: [BUG] Will not run on a Pi 4

Spessmen14, 2023-03-21:
**Describe the bug**
It never captures any handshakes

**To Reproduce**
1. Flash the latest release image to a microsd card
2. Plug in an external wifi dongle (recognized, a name-brand Alfa)
3. sudo service pwnagotchi start
4.  No handshakes ever captured, activity light on wifi dongle blinking

**Expected behavior**
Capturing handshakes

**Environment (please complete the following information):**
Pi4 8GB
4.3inch DSI LCD Display with Case for Raspberry Pi 4B, 800×480 Capacitive Touchscreen MIPI DSI

**Additional context**
Something isn't starting; the web-ui never becomes available.

Logfile paste:

[2023-03-21 20:23:46,236] [INFO] [led] plugin loaded for /sys/class/leds/led0/brightness
[2023-03-21 20:23:46,237] [INFO] Session-stats plugin loaded.
[2023-03-21 20:23:46,246] [WARNING] ui.fps is 0, the display will only update for major changes
[2023-03-21 20:23:46,246] [INFO] initializing waveshare v2 display
[2023-03-21 20:26:14,110] [INFO] grid plugin loaded.
[2023-03-21 20:26:14,110] [INFO] [update] plugin loaded.
[2023-03-21 20:26:14,110] [INFO] [led] plugin loaded for /sys/class/leds/led0/brightness
[2023-03-21 20:26:14,118] [INFO] Session-stats plugin loaded.
[2023-03-21 20:26:14,121] [WARNING] ui.fps is 0, the display will only update for major changes
[2023-03-21 20:26:14,122] [INFO] initializing waveshare v2 display

dj1ch, 2023-07-03:
im pretty sure there is a flag in the config.txt in the boot partition that could disable the onboard wifi. you put this `dtoverlay=pi3-disable-wifi`

josepgl, 2023-08-13:
@Spessmen14 can you give an update?
Did you try the latest version?
Do you see a `mon0` interface?

rsun0525, 2023-08-22:
Since upgrading to 1.7.5, I think this is working now and you can close this issue.