aluminum-ice/pwnagotchi #8: [BUG]Pi0 2 W / Waveshare V3 / Pi Sugar / Constant Reboot

etothex23, 2023-06-03:
**Describe the bug**
Using the current version 1.6.2 image burned to a samsung evo 128gb. Changing the config.toml for bt, passwords, name. Starting the pi zero with an hdmi monitor. The web interface loads, but bettercap does not ever. If the pi does boot it will be only a minute or two and then it will reboot. I am trying to dump the log but it reboots.
**To Reproduce**
Steps to reproduce the behavior:
1. Image disk
2. Copy toml/or not
3. Boot

**Expected behavior**
The system will boot and remain on until purposely rebooted.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version 1.6.2
 - OS version (no change/current)
 - Type of hardware Pi 0 2 W (new), waveshare v3 2.13, pi sugar 2
 - Any additional hardware used hdmi monitor, small for a pi. Keyboard/mouse

**Additional context**
Add any other context about the problem here.
I tried a fan just in case heat was an issue, but i have been running kali, pi os.

glowfishDE, 2023-08-07:
i have the same on a rpi 0 2W with waveshare v3 2.13 and 1.7.6 (latest).

rsun0525, 2023-08-23:
I have this exact setup and working just fine with 1.7.5 (I don't see where 1.7.6 is available).  Raspberry Pi Zero 2W + Pi Sugar 2 + Waveshare v3.  I followed these instructions to add it to 1.7.5, https://github.com/tisboyo/pwnagotchi-pisugar2-plugin, with one exception, after I install the PiSugar Power Manager, I went into localhost:8421 logged in and configured it to always turn on once it has power and also to do the safe shutdown if 1% or less.  After I saw the PiSugar was working and showing properly, then I followed the rest of the instructions, rebooted, and now I can see the battery indicator on pwnagotchi.  

[image]

The only "side effect" is that once pwngatchi plugin was running, going to localhost:8421 doesn't show any valid information anymore.

rsun0525, 2023-08-25:
The one way I've been able to get my device to "reproduce" the crash problem you described by enabling the wrong plugins.  Have you tried disabling all the plugins any plugin you turned on?  Here's what I'm running and so far I've had reasonable stability.

[image]

AnykeyNL, 2023-11-02:
Did you by accident enable ups_lite, because when I did, I got that same behaviour. The unit shutsdown because it reads a low battery status (as I am using PiSugar and not ups_lite