aluminum-ice/pwnagotchi #10: Not scanning APs v 1.7.0

x0r3n, 2023-07-26:
Tried the latest yesterday released 1.7.0 build on my RPI zero 2 W , still not scanning/finding APs :( (same problem was with 1.6.2)
There are No errors in log.

`[2023-07-26 18:56:49,169] [INFO] [ai] ! max_misses_for_recon: 7 -> 6
[2023-07-26 18:56:49,170] [INFO] [ai] ! excited_num_epochs: 6 -> 29
[2023-07-26 18:56:49,170] [INFO] [ai] ! bored_num_epochs: 10 -> 17
[2023-07-26 18:56:49,170] [INFO] [ai] ! sad_num_epochs: 24 -> 19
[2023-07-26 18:56:49,171] [INFO] [ai] ! channels: [4, 10, 11] -> [8, 9, 10, 11]
[2023-07-26 18:57:47,403] [INFO] [epoch 9] duration=00:00:58 slept_for=00:00:55 blind=9 sad=0 bored=0 inactive=10 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=32% mem=60% temperature=47C reward=-0.47000000000000003
[2023-07-26 18:57:47,405] [INFO] [ai] --- training epoch 8/50 ---
[2023-07-26 18:57:47,407] [INFO] [ai] REWARD: -0.470000
[2023-07-26 18:57:47,408] [INFO] [ai] observation:
[2023-07-26 18:57:47,412] [INFO]     aps
[2023-07-26 18:57:47,413] [INFO]     sta
[2023-07-26 18:57:47,414] [INFO]     peers
[2023-07-26 18:57:47,414] [INFO] [ai] saving model to /root/brain.nn ...
[2023-07-26 18:57:47,601] [INFO] [ai] saving /root/brain.json
[2023-07-26 18:57:47,602] [INFO] [ai] worst reward so far: -0.47000000000000003
[2023-07-26 18:57:47,775] [INFO] [ai] setting new policy:
[2023-07-26 18:57:47,780] [INFO] [ai] ! min_rssi: -69 -> -52
[2023-07-26 18:57:47,786] [INFO] [ai] ! ap_ttl: 410 -> 600
[2023-07-26 18:57:47,792] [INFO] [ai] ! sta_ttl: 225 -> 232
[2023-07-26 18:57:47,797] [INFO] [ai] ! recon_time: 55 -> 31
[2023-07-26 18:57:47,803] [INFO] [ai] ! max_inactive_scale: 8 -> 10
[2023-07-26 18:57:47,808] [INFO] [ai] ! recon_inactive_multiplier: 1 -> 3
[2023-07-26 18:57:47,814] [INFO] [ai] ! hop_recon_time: 44 -> 53
[2023-07-26 18:57:47,819] [INFO] [ai] ! min_recon_time: 7 -> 22
[2023-07-26 18:57:47,825] [INFO] [ai] ! max_interactions: 20 -> 25
[2023-07-26 18:57:47,831] [INFO] [ai] ! max_misses_for_recon: 6 -> 9
[2023-07-26 18:57:47,837] [INFO] [ai] ! excited_num_epochs: 29 -> 12
[2023-07-26 18:57:47,847] [INFO] [ai] ! bored_num_epochs: 17 -> 11
[2023-07-26 18:57:47,853] [INFO] [ai] ! sad_num_epochs: 19 -> 6
[2023-07-26 18:57:47,859] [INFO] [ai] ! channels: [8, 9, 10, 11] -> [1, 2, 4, 8]`

This is my interface list 
`lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 750  bytes 1275079 (1.2 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 750  bytes 1275079 (1.2 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

mon0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        unspec E4-5F-01-7E-90-EE-30-30-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.2  netmask 255.255.255.0  broadcast 10.0.0.255
        inet6 fe80::fc51:11ff:fecf:5623  prefixlen 64  scopeid 0x20<link>
        ether fe:51:11:cf:56:23  txqueuelen 1000  (Ethernet)
        RX packets 1129  bytes 128936 (125.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1164  bytes 1002786 (979.2 KiB)
        TX errors 0  dropped 3 overruns 0  carrier 0  collisions 0`

x0r3n, 2023-07-26:
In my case I found the problem.
My raspberry zero 2 W using brcmfmac43430-sdio firmware and not brcmfmac43436-sdio which was patched in this build.
You can see it at boot which firmware is loading for your chip. You can check it with command 
`dmesg | grep brcmf_fw_alloc_request`
in my case:
[ 6.174867] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1

jayofelony, 2023-07-27:
I also have a Pi 0 2W with the Pi3B+ chip.
```
[    9.173508] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    9.260619] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436s-sdio for chip BCM43430/1
[    9.299803] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43436s-sdio.clm_blob failed with error -2
[    9.344957] usbcore: registered new interface driver brcmfmac
[    9.524098] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    9.525185] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jul  5 2022 21:29:56 version 7.45.96.141 (g53ba0094) FWID 01-eef9b3bf es7`
```

Tried loading both firmwares for the 43430, both are not being loaded at boot after depmod -a.

renewedbullet, 2023-07-27:
same issue as above 43430/1

jayofelony, 2023-07-27:
I have one working image for the pi02w, but I think it's an arm (32bit) build. I just flashed it and will see what driver it is using.

jayofelony, 2023-07-27:
Yes but it is the same pi that will not load the drivers on 64 bit raspios. But for some reason it will load this arm build driver.

renewedbullet, 2023-07-28:
Appreciate the effort ill reuse this zero 2 for some other project

pi@pwnagotchi:~ $ sudo dmesg | grep brcmf_fw_alloc_request
[    6.484808] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.868646] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1

x0r3n, 2023-07-29:
1.7.3 looks work, finding APs. I will test it longer time now.
I am very grateful for your work! Thank you for your invested time to it.

x0r3n, 2023-07-29:
> 1.7.3 looks work, finding APs. I will test it longer time now. I am very grateful for your work! Thank you for your invested time to it.
Atm looks good :)
[image]