aluminum-ice/pwnagotchi #15: [BUG] RPiZeroW no device mon0

Evanito, 2023-07-31:
**Describe the bug**
When installed on the RPi Zero W (2017 aka 1), no wireless device is found. The output of `ip a` only shows loopback and usb.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash 1.7.3 (most recent at this time)
2. Run on a RPiZeroW
3. bettercap will fail to launch, the rest of the process operates as normal. pwnagotchi waits infinitely while bettercap attempts to launch

**Expected behavior**
mon0 to be present 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version
   - 1.7.3
 - OS version
   - ?
 - Type of hardware
   - RPi Zero W, 1st edition
 - Any additional hardware used

**Additional context**
Add any other context about the problem here.

BarnOwlDown, 2023-07-31:
I have the same problem. This is what I found in kern.log:

```
May  3 03:17:43 pwnagotchi kernel: [    0.150962] raspberrypi-firmware soc:firmware: Attached to firmware from 2023-05-09T12:18:12, variant start_cd
May  3 03:17:43 pwnagotchi kernel: [    0.160987] raspberrypi-firmware soc:firmware: Firmware hash is 30aa0d70ab280427ba04ebc718c81d4350b9d394
```

```
Jul 31 18:04:24 BarnOwl2 kernel: [   20.680302] cfg80211: Loading compiled-in X.509 certificates for regulatory database
Jul 31 18:04:24 BarnOwl2 kernel: [   21.128797] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
Jul 31 18:04:24 BarnOwl2 kernel: [   21.300220] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
Jul 31 18:04:24 BarnOwl2 kernel: [   22.071914] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
Jul 31 18:04:24 BarnOwl2 kernel: [   22.230911] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
Jul 31 18:04:24 BarnOwl2 kernel: [   22.236916] usbcore: registered new interface driver brcmfmac
Jul 31 18:04:24 BarnOwl2 kernel: [   22.702136] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
Jul 31 18:04:24 BarnOwl2 kernel: [   22.715202] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Jul 31 18:04:24 BarnOwl2 kernel: [   22.843301] ------------[ cut here ]------------
Jul 31 18:04:24 BarnOwl2 kernel: [   22.843369] WARNING: CPU: 0 PID: 111 at kernel/workqueue.c:1486 __queue_work+0x2e4/0x3a4
Jul 31 18:04:24 BarnOwl2 kernel: [   22.843378] Modules linked in: brcmfmac brcmutil sha256_generic libsha256 cfg80211 raspberrypi_hwmon rfkill i2c_bcm2835 spi_bcm2835aux bcm2835_codec(C) snd_bcm2835(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem videobuf2_vmalloc snd_pcm bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_memops snd_timer videobuf2_v4l2 videobuf2_common snd videodev vc_sm_cma(C) mc spi_bcm2835 uio_pdrv_genirq uio fixed i2c_dev usb_f_ecm g_ether usb_f_rndis u_ether libcomposite dwc2 udc_core roles ip_tables x_tables ipv6
Jul 31 18:04:24 BarnOwl2 kernel: [   22.843641] CPU: 0 PID: 111 Comm: kworker/u2:2 Tainted: G         C        5.10.103+ #1529
Jul 31 18:04:24 BarnOwl2 kernel: [   22.843649] Hardware name: BCM2835
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844086] Workqueue: brcmf_wq/mmc1:0001:1 brcmf_sdio_dataworker [brcmfmac]
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844105] Backtrace: 
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844161] [<c087714c>] (dump_backtrace) from [<c08774cc>] (show_stack+0x20/0x24)
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844181]  r7:000005ce r6:00000009 r5:c003c064 r4:c0a73108
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844208] [<c08774ac>] (show_stack) from [<c087a740>] (dump_stack+0x28/0x30)
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844239] [<c087a718>] (dump_stack) from [<c0023164>] (__warn+0xe8/0x110)
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844252]  r5:c003c064 r4:c0a6b130
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844272] [<c002307c>] (__warn) from [<c0877ae0>] (warn_slowpath_fmt+0x68/0xd8)
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844288]  r7:00000009 r6:c003c064 r5:000005ce r4:c0a6b130
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844314] [<c0877a7c>] (warn_slowpath_fmt) from [<c003c064>] (__queue_work+0x2e4/0x3a4)
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844335]  r9:c102f000 r8:c0d00a54 r7:00000001 r6:000000f8 r5:defb8300 r4:c395acac
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844355] [<c003bd80>] (__queue_work) from [<c003c1cc>] (queue_work_on+0xa8/0xb0)
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844375]  r10:c33f0610 r9:00000000 r8:c102f000 r7:00000001 r6:00000000 r5:c395acac
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844384]  r4:60000113
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844668] [<c003c124>] (queue_work_on) from [<bf6722ac>] (brcmf_fw_crashed+0x54/0x64 [brcmfmac])
Jul 31 18:04:24 BarnOwl2 kernel: [   22.844692]  r9:00000000 r8:c33f0610 r7:c19d9200 r6:c4d00960 r5:c19da408 r4:c39584c0
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845213] [<bf672258>] (brcmf_fw_crashed [brcmfmac]) from [<bf66a9b0>] (brcmf_sdio_dataworker+0x640/0x2480 [brcmfmac])
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845228]  r5:00040012 r4:c33f0c00
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845510] [<bf66a370>] (brcmf_sdio_dataworker [brcmfmac]) from [<c003dd84>] (process_one_work+0x208/0x4dc)
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845534]  r10:c33f0614 r9:00000000 r8:00000000 r7:c19d9200 r6:00000040 r5:c3052240
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845544]  r4:c33f0610
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845567] [<c003db7c>] (process_one_work) from [<c003e08c>] (worker_thread+0x34/0x594)
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845589]  r10:c1024200 r9:00000088 r8:c0c2bea0 r7:c1024214 r6:c3052254 r5:c1024200
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845598]  r4:c3052240
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845622] [<c003e058>] (worker_thread) from [<c00439c8>] (kthread+0x148/0x15c)
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845642]  r10:c112de88 r9:c3052240 r8:c003e058 r7:c30f8000 r6:00000000 r5:c3094980
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845654]  r4:c2081140 r3:00000000
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845674] [<c0043880>] (kthread) from [<c00082ec>] (ret_from_fork+0x14/0x28)
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845685] Exception stack(0xc30f9fb0 to 0xc30f9ff8)
Jul 31 18:04:24 BarnOwl2 kernel: [   22.845699] 9fa0:                                     00000000 00000000 00000000 00000000
[... cut ...]
Jul 31 18:04:24 BarnOwl2 kernel: [   27.856858] brcmfmac: dongle trap info: type 0xc @ epc 0x0001f106
Jul 31 18:04:24 BarnOwl2 kernel: [   27.856858]   cpsr 0x0000000c spsr 0x01000010 sp 0x0006fb04
Jul 31 18:04:24 BarnOwl2 kernel: [   27.856858]   lr   0x0001f0fd pc   0x0001f106 offset 0x6fab0
Jul 31 18:04:24 BarnOwl2 kernel: [   27.856858]   r0   0x0003a6a0 r1   0x00082540 r2 0x00000054 r3 0x00002095
Jul 31 18:04:24 BarnOwl2 kernel: [   27.856858]   r4   0x0006fb50 r5   0x000683f0 r6 0x00000001 r7 0x0000123d
```

SpeedAndPower, 2023-08-01:
same issue here, Both:
/lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.clm_blob
/lib/firmware/brcm/brcmfmac43430-sdio.clm_blob
Have been removed by ansible.

`May  3 03:48:36 pwnagotchi kernel: [   18.651369] videodev: Linux video capture interface: v2.00
May  3 03:48:36 pwnagotchi kernel: [   18.790998] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
May  3 03:48:36 pwnagotchi kernel: [   18.805611] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
May  3 03:48:36 pwnagotchi kernel: [   18.825170] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
May  3 03:48:36 pwnagotchi kernel: [   18.845968] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
May  3 03:48:36 pwnagotchi kernel: [   18.892321] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
May  3 03:48:36 pwnagotchi kernel: [   18.922923] bcm2835_audio bcm2835_audio: there is not valid maps for state default
May  3 03:48:36 pwnagotchi kernel: [   18.980855] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
May  3 03:48:36 pwnagotchi kernel: [   18.989452] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
May  3 03:48:36 pwnagotchi kernel: [   19.016382] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
May  3 03:48:36 pwnagotchi kernel: [   19.061011] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
May  3 03:48:36 pwnagotchi kernel: [   20.867014] cfg80211: Loading compiled-in X.509 certificates for regulatory database
May  3 03:48:36 pwnagotchi kernel: [   21.336262] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
May  3 03:48:36 pwnagotchi kernel: [   21.368061] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
May  3 03:48:36 pwnagotchi kernel: [   22.070925] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
May  3 03:48:36 pwnagotchi kernel: [   22.270892] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
May  3 03:48:36 pwnagotchi kernel: [   22.278146] usbcore: registered new interface driver brcmfmac
May  3 03:48:36 pwnagotchi kernel: [   22.728390] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
May  3 03:48:36 pwnagotchi kernel: [   22.748750] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
May  3 03:48:36 pwnagotchi kernel: [   22.893272] ------------[ cut here ]------------
May  3 03:48:36 pwnagotchi kernel: [   22.893339] WARNING: CPU: 0 PID: 111 at kernel/workqueue.c:1486 __queue_work+0x2e4/0x3a4
May  3 03:48:36 pwnagotchi kernel: [   22.893350] Modules linked in: brcmfmac brcmutil sha256_generic libsha256 cfg80211 rfkill raspberrypi_hwmon spi_bcm2835aux i2c_bcm2835 bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem snd_bcm2835(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops spi_bcm2835 videobuf2_v4l2 videobuf2_common snd_pcm snd_timer snd videodev mc vc_sm_cma(C) fixed uio_pdrv_genirq uio i2c_dev usb_f_ecm g_ether usb_f_rndis u_ether libcomposite dwc2 udc_core roles ip_tables x_tables ipv6
May  3 03:48:36 pwnagotchi kernel: [   22.893611] CPU: 0 PID: 111 Comm: kworker/u2:2 Tainted: G         C        5.10.103+ #1529
May  3 03:48:36 pwnagotchi kernel: [   22.893619] Hardware name: BCM2835
May  3 03:48:36 pwnagotchi kernel: [   22.894053] Workqueue: brcmf_wq/mmc1:0001:1 brcmf_sdio_dataworker [brcmfmac]
May  3 03:48:36 pwnagotchi kernel: [   22.894072] Backtrace:
May  3 03:48:36 pwnagotchi kernel: [   22.894130] [<c087714c>] (dump_backtrace) from [<c08774cc>] (show_stack+0x20/0x24)
May  3 03:48:36 pwnagotchi kernel: [   22.894151]  r7:000005ce r6:00000009 r5:c003c064 r4:c0a73108
May  3 03:48:36 pwnagotchi kernel: [   22.894177] [<c08774ac>] (show_stack) from [<c087a740>] (dump_stack+0x28/0x30)
May  3 03:48:36 pwnagotchi kernel: [   22.894208] [<c087a718>] (dump_stack) from [<c0023164>] (__warn+0xe8/0x110)
May  3 03:48:36 pwnagotchi kernel: [   22.894221]  r5:c003c064 r4:c0a6b130
May  3 03:48:36 pwnagotchi kernel: [   22.894240] [<c002307c>] (__warn) from [<c0877ae0>] (warn_slowpath_fmt+0x68/0xd8)
May  3 03:48:36 pwnagotchi kernel: [   22.894256]  r7:00000009 r6:c003c064 r5:000005ce r4:c0a6b130
May  3 03:48:36 pwnagotchi kernel: [   22.894281] [<c0877a7c>] (warn_slowpath_fmt) from [<c003c064>] (__queue_work+0x2e4/0x3a4)
May  3 03:48:36 pwnagotchi kernel: [   22.894301]  r9:c102f000 r8:c0d00a54 r7:00000001 r6:0000014c r5:defb8300 r4:c45aacac
May  3 03:48:36 pwnagotchi kernel: [   22.894321] [<c003bd80>] (__queue_work) from [<c003c1cc>] (queue_work_on+0xa8/0xb0)
May  3 03:48:36 pwnagotchi kernel: [   22.894342]  r10:c31bca10 r9:00000000 r8:c102f000 r7:00000001 r6:00000000 r5:c45aacac
May  3 03:48:36 pwnagotchi kernel: [   22.894351]  r4:60000113
May  3 03:48:36 pwnagotchi kernel: [   22.894636] [<c003c124>] (queue_work_on) from [<bf6332ac>] (brcmf_fw_crashed+0x54/0x64 [brcmfmac])
May  3 03:48:36 pwnagotchi kernel: [   22.894659]  r9:00000000 r8:c31bca10 r7:c19d8200 r6:c469a680 r5:c19d9408 r4:c45a84c0
May  3 03:48:36 pwnagotchi kernel: [   22.895176] [<bf633258>] (brcmf_fw_crashed [brcmfmac]) from [<bf62b9b0>] (brcmf_sdio_dataworker+0x640/0x2480 [brcmfmac])
[... cut ...]
May  3 03:48:36 pwnagotchi kernel: [   22.895679] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
May  3 03:48:36 pwnagotchi kernel: [   22.895695] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
May  3 03:48:36 pwnagotchi kernel: [   22.895714]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0043880
May  3 03:48:36 pwnagotchi kernel: [   22.895723]  r4:c304e680
May  3 03:48:36 pwnagotchi kernel: [   22.895733] ---[ end trace b63e2e5b1bb43148 ]---
`

SpeedAndPower, 2023-08-01:
FWIW, Removing  /lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.clm_blob  allowed it to come up with no crash.

I thought 43430 firmware has a built in CLM database so odd its complaining about a blob missing. But still no Mon0 interface so bettercap loops

```
pi@pwnagotchi:~ $ dmesg | grep brcm
[   22.145576] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   22.393545] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   22.424141] usbcore: registered new interface driver brcmfmac
[   22.867314] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   22.867621] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   22.867796] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   22.869144] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[   33.701303] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
pi@pwnagotchi:~ $ iwconfig
lo        no wireless extensions.

usb0      no wireless extensions.

wlan0     IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off

```

SpeedAndPower, 2023-08-02:
It still ends up not ever launching a mon0 interface so bettercap doesn't launch. Im building a few different builds right now for testing to see what is up. I don't have access to anything but three of my space Pi Zero W( not the 2). Ill update tomorrow EOD on if I find anything

MonstaB, 2023-08-06:
Installed latest 1.7.4 on original pi zero w.
Had same issues as everyone did a fresh install on sdcard.
Deleted blob file,
Cloned nexmon.
Built nexmon for pizero 
Installed new driver I made.
Did iwconfig had mon0 and wlan0
Stopped here.
Reboot pi not attached to PC.
Started in manual mode( I didn’t put that other file in earlier that was mentioned in another post.)
Turned off plugged in pc , turned on,
Decided to check iwconfig first and mon0 s gone.
Dunno if this helps but I’ll do it all again tonight and build the nexmon tools too.

MonstaB, 2023-08-07:
this is where im at after building nexmon, after a reboot its back to no mon0    

                       !!! WARNING !!!
    Our software may damage your hardware and may void your hardware’s
     warranty! You use our tools at your own risk and responsibility

  COLLECTING STATISTICS read /home/pi/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43430-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 5.10 brcmfmac_5.10.y-nexmon/brcmfmac.ko (details: log/driver.log)
  COPYING brcmfmac43430-sdio.bin => /lib/firmware/brcm/brcmfmac43430-sdio.bin
  UNLOADING brcmfmac
  RELOADING brcmfmac
root@pwnagotchi:/home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon# iwconfig
lo        no wireless extensions.

usb0      no wireless extensions.

wlan0     IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off

mon0      IEEE 802.11  Mode:Monitor  Frequency:2.412 GHz  Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on

root@pwnagotchi:/home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon# su pi
pi@pwnagotchi:~/nexmon/patches/bcm43430a1/7_45_41_46/nexmon $ dmesg | grep brcm
[   21.986789] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   22.387550] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   22.395097] usbcore: registered new interface driver brcmfmac
[   22.837981] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   22.838286] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   22.838468] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   22.839811] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[   31.720784] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[  315.244560] usbcore: deregistering interface driver brcmfmac
[  315.717037] brcmfmac: loading out-of-tree module taints kernel.
[  315.756348] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[  315.792518] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[  315.792883] usbcore: registered new interface driver brcmfmac
[  316.174453] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[  316.174582] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[  316.175665] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[  317.196322] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[  317.196350] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[  322.895895] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[  322.895923] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[  323.019792] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[  323.019831] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[  323.019845] brcmfmac: brcmf_mon_add_vif: Adding vif "mon0"
pi@pwnagotchi:~/nexmon/patches/bcm43430a1/7_45_41_46/nexmon $

MonstaB, 2023-08-13:
ok, First step fresh install of 1.7.4
change config.toml to vaveshare_3
fix no auto reboot by fixing line 48 of pwnlib file "if grep -qi 'up' /sys/class/net/$1/operstate; then"  Thanks to those involved.
then restart
a "dmesg | grep brcm " still shows issues that deleting the blob files will fix but i'm not sure if this is a solution or just symtem of another issue.
now the issue ive found running a "modinfo brcmfmac" shows we are loading from "/lib/modules/5.10.103+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko" 
but
in your /builder/pwnagotchi.yml
"- name: copy modified driver
    copy:
      src: /usr/local/src/nexmon/patches/driver/brcmfmac_5.10.y-nexmon/brcmfmac.ko
      dest: /lib/modules/5.10.103-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko"
which is the wrong location for the RPiZeroW ,
i'm assuming the multicore or the v2 boards uses a different kernal.
I think this is the right src just wrong dest in our case i just dont know how to adjust the code to account for this because just changing that line will probably break everyone else.

workaround for now
build nexmon recopy files into correct location.
testing now

Dviros, 2023-08-20:
Same issue here :(
Any way to share the blob file?

Dviros, 2023-08-20:
Damn! That was painful to fix:
[image]

So, steps to get this done (make sure to drink a lot and getting a large size SD card):
```bash
sudo apt install raspberrypi-kernel-headers git libgmp3-dev gawk
git clone https://github.com/seemoo-lab/nexmon.git
cd nexmon
sudo -i
source setup_env.sh
make
cd patches/bcm43430a1/7_45_41_46/nexmon
make
make install-firmware
cd ../../../../utilities/libnexio/
make
cd ../nexutil/
make
make install
ifconfig wlan0 down
nexutil -m2
ifconfig wlan0 up
```

MonstaB, 2023-08-20:
Nice work,

I was asking about the blob but 1.5.5 doesn't have that blob either so it's not needed at all I guess.

Was this on the v1 or v2 board just to clarify?

Did you have to change the . ko file or just using the original one???,  is it loading from the v7 kernel or or the other

I'll try your last 3 steps tonight and see what happenes

Dviros, 2023-08-20:
> Nice work,
>
> I was asking about the blob but 1.5.5 doesn't have that blob either so it's not needed at all I guess.
>
> Was this on the v1 or v2 board just to clarify?
>
> Did you have to change the . ko [...]

Just edited my reply, added perssistency.
Think I'm using the v1 board.

MonstaB, 2023-08-20:
Ok. Thanks.
My issue is just that wlan0 stays up so hopefully stay will be sorted tonight when I try, then on to 1.7.5 see if it works there

MonstaB, 2023-08-21:
> 
> 
> To make sure the driver is perssistent between reboots:
> 
> ```bash
> 
> driver=$(modinfo brcmfmac | head -n 1)
> 
> driver=$(echo $driver | sed 's/^.\{'10'\}//')
> 
> mv $driver $driver.orig
> 
> driver_path=${driver%???????????}
> 
> cp /home/pi/nexmon/patches/driver/brcmfmac_5.10.y-nexmon/brcmfmac.ko $driver_path
> 
> depmod -a
> 
> ```

Just so I am clear this command copies the file too /lib/modules/5.10.103+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/
And not where the builder copies it?

Dviros, 2023-08-21:
Only the .ko file needs replacing to be perssistent. @aluminum-ice I'll take a look at the ansible script and reply

Dviros, 2023-08-21:
@aluminum-ice think I found the issue:
[image]

The script copies the driver to this path
```/lib/modules/5.10.103-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko```
Instead of
```/lib/modules/5.10.103+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko```
Mind the -v7+

MonstaB, 2023-08-21:
> @aluminum-ice think I found the issue:
>
> [image]
>
>
>
> The script copies the driver to this path
>
> ```/lib/modules/5.10.103-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko``` [...]

If we change just that line it will break everyone else that is allready working, that's why I suggested  adding a copy to the correct place as well when the driver is made, I believe the same driver on the RPiZero2w goes where it is now it's just the RPiZeroW that uses the 5.10.103+

NeonLightning, 2023-08-22:
> Damn! That was painful to fix: [image]
> 
> So, steps to get this done (make sure to drink a lot and getting a large size SD card):
> 
> ```shell
> sudo apt install raspberrypi-kernel-headers git libgmp3-dev gawk
> git clone https://github.com/seemoo-lab/nexmon.git
> cd nexmon
> sudo -i
> source setup_env.sh
> make
> cd patches/bcm43430a1/7_45_41_46/nexmon
> make
> make install-firmware
> cd ../../../../utilities/libnexio/
> make
> cd ../nexutil/
> make
> make install
> ifconfig wlan0 down
> nexutil -m2
> ifconfig wlan0 up
> ```
> 
> To make sure the driver is perssistent between reboots:
> 
> ```shell
> driver=$(modinfo brcmfmac | head -n 1)
> driver=$(echo $driver | sed 's/^.\{'10'\}//')
> mv $driver $driver.orig
> driver_path=${driver%???????????}
> cp /home/pi/nexmon/patches/driver/brcmfmac_5.10.y-nexmon/brcmfmac.ko $driver_path
> depmod -a
> ```

when i tried that i still get
[2023-08-23 00:57:27,983] [INFO] pwnagotchi@2f7983c480bed71fea48ea83d85eb276b8ac072ff6ba0d75d86cdd986a70ef82 (v1.7.5)
[2023-08-23 00:57:27,989] [INFO] entering auto mode ...
[2023-08-23 00:57:28,017] [INFO] [ai] bootstrapping dependencies ...
[2023-08-23 00:57:29,207] [INFO] connecting to http://localhost:8081/api ...
[2023-08-23 00:57:30,589] [INFO] web ui available at http://0.0.0.0:8080/
[2023-08-23 00:57:38,907] [INFO] found monitor interface: mon0
[2023-08-23 00:57:38,962] [INFO] supported channels: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14]
[2023-08-23 00:57:39,111] [INFO] handshakes will be collected inside /root/handshakes
[2023-08-23 00:57:47,691] [INFO] error 400: error while initializing mon0 to channel 1: iw: out=command failed: Device or resource busy (-16) err=exit status 240
and then it goes into manu again.

NeonLightning, 2023-08-30:
tried again on the newest image and those steps to fix it are working
5.10.103+

MonstaB, 2023-08-30:
> @NeonLightning @MonstaB Can you please tell me what you see when you run 'uname -r' on your RPiZW?

yeah mate i get  
pi@pwnagotchi:~ $ uname -r
5.10.103+

MonstaB, 2023-09-01:
> Try [v1.7.6](https://github.com/aluminum-ice/pwnagotchi/releases/tag/v1.7.6)

I'll do it now I'll let you know in an hour or 2

C4rdsh4rk, 2023-09-01:
same problem here on pi0 and 1.7.6

C4rdsh4rk, 2023-09-01:
I'm trying to build everything now from source, but it's taking forever. I'm also trying to speed things up, by building in a arm vm, but I'm not sure if the chipset must be virtualized too, to compile nexmon's patch correctly

C4rdsh4rk, 2023-09-01:
Any tips? I'm following the instructions on the nexmon git, but I'm not sure how to see if it worked

C4rdsh4rk, 2023-09-01:
Alright, I'll order one^^ but I'd still like to learn how to make it work though

C4rdsh4rk, 2023-09-01:
I refunded my piz2w because it had the old chipset and I couldn't get that to work.
About one day after the refund, you added support for both chipsets xD
But I'd understand abandoning the pizw. It's not fun and runs at 100% all the time

NeonLightning, 2023-09-01:
ah shame.  i can't afford any more electronics :( but i get it. maintaining many things at once gets complicated and frustrating.

MonstaB, 2023-09-01:
I just wanna get one version of this with no kali Linux working on the v1 board, the issue is at the moment I can get v1 boards but not any other pi boards at the moment. Otherwise I would move to a zero2.

I tried making an image of my fork on my pi but won't compile on it, did it in a kali vm dunno if that's correct or not. Ended up being 9gig so don't know if that seems correct.