aluminum-ice/pwnagotchi #16: Not booting into AUTO mode [BUG]

azzarin, 2023-08-03:
The pi does not boot to AUTO mode when usb is plugged into power only interface. It only boots to MANU mode. 
I can ssh to the pi and restart into AUTO mode and it works just fine. But i have not set up BT connection, so on the go its not possible. 

**To Reproduce**
Steps to reproduce the behavior:
1. ... Plug usb to power USB port.

**Expected behavior**
Boot to AUTO mode not MANU. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version V1.7.4
 - OS version
 - Type of hardware Pi zero 2 W
 - Any additional hardware used: Inky screen

**Additional context**
Add any other context about the problem here.

Jilten, 2023-08-05:
This is happening for me as well :(
Pwnagotchi version V1.7.4
Type of hardware Pi zero 2 W
Any additional hardware used: Waveshare 2

glowfishDE, 2023-08-07:
i also have the same on 1.7.6. A quick solution for me was to do this:

1) touch /root/.pwnagotchi-auto
2) chattr +i /root/.pwnagotchi-auto
3) reboot

after that it will always reboot in AUTO mode. The 2nd command will prevent the OS to delete the /root/.pwnagotchi-auto file on startup.

Coach4, 2023-08-07:
I am having this same issue with 1.7.4.  Will this work?  Also is this going into powershell or into the SSH?  Thanks!  Newbie here just looking for some help.   Thanks again!

i also have the same on 1.7.6. A quick solution for me was to do this:

touch /root/.pwnagotchi-auto
chattr +i /root/.pwnagotchi-auto
reboot
after that it will always reboot in AUTO mode. The 2nd command will prevent the OS to delete the /root/.pwnagotchi-auto file on startup.

glowfishDE, 2023-08-07:
hi @Coach4 - you have to issue these commands when you are connected to the pwnagotchi via SSH. Powershell is only for windows. You have to connect via ssh pi@ip.ip.ip.ip and than you can issue the commands above.

Coach4, 2023-08-08:
@glowfishDE.  Just wanted to say thanks!  Works absolutely amazing!

ZexX-pt, 2023-08-09:
Hi too am having the same issue.
The work around provided by @Coach4 fixes for now, but it is not ideal.

azzarin, 2023-08-09:
> I am having this same issue with 1.7.4. Will this work? Also is this going into powershell or into the SSH? Thanks! Newbie here just looking for some help. Thanks again!
>
> i also have the same on [...]

This makes perfect sense and is described in the official documentation.

glowfishDE, 2023-08-09:
> Hi too am having the same issue. The work around provided by @Coach4 fixes for now, 

work around was provided by @glowfishDE ;) just sayin´

connornorton, 2023-08-11:
pull request submitted fixing this issue #19