aluminum-ice/pwnagotchi #17: PI Zero 2 W - No WiFi

CBTux, 2023-08-04:
No WiFi

During the boot process, WiFi firmware not loaded, at least in my case

**Environment (please complete the following information):**
 - Pwnagotchi V1.7.4
 - PI Zero 2 W

dmesg | grep brcmfmac
[    6.249994] brcmfmac: loading out-of-tree module taints kernel.
[    6.278710] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    6.292902] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.293226] usbcore: registered new interface driver brcmfmac
**[    6.322918] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-2-w.txt failed with error -2**
[    6.667093] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.667272] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    6.668251] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[    8.574321] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[    8.574354] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[    9.709728] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[    9.709753] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[    9.801349] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[    9.801394] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[    9.801414] brcmfmac: brcmf_mon_add_vif: Adding vif "mon0"

ZexX-pt, 2023-08-09:
Hi am also seeing the same errors.
My Pi Zero 2W also has the same wifi chip 43430/1.

MonstaB, 2023-08-12:
can one of you guys run this and tell me the output..
modinfo brcmfmac

ZexX-pt, 2023-08-12:
I will try help with the compilation, but in the meantime here are the logs.

[dmesg.log](https://github.com/aluminum-ice/pwnagotchi/files/12328399/dmesg.log)
[modinfo_brcmfmac.log](https://github.com/aluminum-ice/pwnagotchi/files/12328400/modinfo_brcmfmac.log)

MonstaB, 2023-08-13:
Look I'll be honest Im only new to GitHub and don't really know how pull requests and stuff work so I won't do that as yet, I've identified the issue and my way of fixing it would probably break everything that isn't the PizeroW chipset, I can get it running now but the current workaround for auto mode breaks my fix for the chipset, any idea how to get the auto detect on the port to work like the original?  I'll write my workaround in the other thread in an hour or two

MonstaB, 2023-08-13:
Try doing a ifconfig,
If you have both a wlan0 and a mon0
Run 
sudo ifconfig wlan0 down

See what that does

ZexX-pt, 2023-08-13:
I do have both interfaces, wlan0 and mon0, but only mon0 is up.
The interface wlan0 is already down.

rsun0525, 2023-08-22:
I installed 1.7.5 and the output now looks like this, so I think it's working now and can close this ticket?

dmesg | grep brcmfmac
[    6.321289] brcmfmac: loading out-of-tree module taints kernel.
[    6.349094] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    6.355559] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.355892] usbcore: registered new interface driver brcmfmac
[    6.755698] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.755862] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    6.756801] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[    8.553194] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[    8.553218] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[    9.442936] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[    9.442959] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[ 3452.562644] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[ 3452.562657] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled

ZexX-pt, 2023-08-22:
Hi,

That is weird. Version 1.7.5 did not fix it for me.

pi@pwnagotchi85:~ $ pwnagotchi --version
1.7.5
pi@pwnagotchi85:~ $ dmesg | grep brcmfmac
[    7.998521] brcmfmac: loading out-of-tree module taints kernel.
[    8.027990] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    8.038073] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    8.038380] usbcore: registered new interface driver brcmfmac
[    8.068119] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-2-w.txt failed with error -2
[    8.430703] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    8.430923] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    8.431866] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[    9.954689] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[    9.954714] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   11.226901] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[   11.226924] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   11.309550] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[   11.309584] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[   11.309598] brcmfmac: brcmf_mon_add_vif: Adding vif "mon0"

rsun0525, 2023-08-22:
Seems like there's more than one hardware version of Raspberry Pi Zero 2 W :-(

https://github.com/RPi-Distro/firmware-nonfree/issues/23

rsun0525, 2023-08-22:
Did you also run all the updates on the Raspberry Pi Zero 2W?

sudo apt update
sudo apt full-upgrade -y
sudo apt autoremove -y
sudo reboot

ZexX-pt, 2023-08-22:
I though that upgrading some version of a package could break pwnagotchi so I did not do it.

However, I shared me connection with pwnagotchi usb ethernet and have the upgrade and I get the same error.

Do I have to upgrade the kernel or any driver?

pi@pwnagotchi85:~ $ uname -a
Linux pwnagotchi85 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux

rsun0525, 2023-08-22:
I also connect my Raspberry Pi Zero 2 W through Ethernet as well through the USB port (and often keep it that way while running it) and haven't had any issues since upgrading to 1.7.5, I don't make any additional changes at the OS level other than upgrading all the packages, so my only guess is that your Raspberry Pi Zero 2 W is a different hardware version.

rsun0525, 2023-08-29:
I guess I'm lucky :-)

 brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2

turcane, 2023-08-30:
I can confirm. It's working :)

ZexX-pt, 2023-09-01:
Hi,

Sorry for the delay, but I only could test it today.

I don't see any improvement. Same errors as before:

pi@pwnagotchi85:~ $ dmesg | grep brcmfmac
[    6.147501] brcmfmac: loading out-of-tree module taints kernel.
[    6.187518] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    6.193679] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.193986] usbcore: registered new interface driver brcmfmac
[    6.227439] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-2-w.txt failed with error -2
[    6.578607] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.578763] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    6.579821] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[    8.161323] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[    8.161358] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[    9.438004] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[    9.438032] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[    9.529245] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[    9.529279] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[    9.529293] brcmfmac: brcmf_mon_add_vif: Adding vif "mon0"

pi@pwnagotchi85:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.103-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1529 SMP Tue Mar 8 12:21:37 GMT 2022
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero 2 W Rev 1.0
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x1ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001effffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001effffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001effffff]
[    0.000000] On node 0 totalpages: 126976
[    0.000000]   DMA zone: 1116 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 126976 pages, LIFO batch:31
[    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
[    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 125860
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:92:1E:FE vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 fsck.repair=yes rootwait modules-load=dwc2,g_ether
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 420412K/507904K available (10240K kernel code, 1312K rwdata, 2952K rodata, 1024K init, 862K bss, 21956K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 32081 entries in 95 pages
[    0.000000] ftrace: allocated 94 pages with 5 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x3ac/0x580 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000024] Switching to timer-based delay loop, resolution 52ns
[    0.000305] Console: colour dummy device 80x30
[    0.001142] printk: console [tty1] enabled
[    0.001212] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[... 217 lines cut ...]
[    9.429420] fuse: init (API version 7.32)
[    9.438004] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[    9.438032] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[    9.529245] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[... cut ...]
[   14.901503] Bluetooth: HCI UART protocol H4 registered
[   14.901593] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   14.901797] Bluetooth: HCI UART protocol Broadcom registered
[   15.345761] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.345776] Bluetooth: BNEP filters: protocol multicast
[   15.345797] Bluetooth: BNEP socket layer initialized
[   23.814537] device mon0 entered promiscuous mode

The reported version also seems a bit off:
pi@pwnagotchi85:~ $ pwnagotchi --version
1.7.51

turcane, 2023-09-01:
I can provide later this day dmesg results. I have just checked mon0 interface and scanning with mon0 device, which was working.

turcane, 2023-09-01:
`root@pwnny-cage:~# dmesg | grep brcmfmac
[    6.586340] brcmfmac: loading out-of-tree module taints kernel.
[    6.586346] brcmfmac: loading out-of-tree module taints kernel.
[    6.669865] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    6.679052] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.679335] usbcore: registered new interface driver brcmfmac
[    6.708043] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-2-w.txt failed with error -2
[    7.068258] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    7.068406] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    7.069397] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Aug  7 2017 00:46:29 version 7.45.41.46 (r666254 CY) FWID 01-f8a78378
[    8.757260] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[    8.757286] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[    9.924855] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[    9.924878] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   10.013097] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[   10.013132] brcmfmac: brcmf_mon_add_vif: brcmf_mon_add_vif called
[   10.013146] brcmfmac: brcmf_mon_add_vif: Adding vif "mon0"
[   38.969449] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
[   38.969465] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   39.011364] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
[   39.011381] brcmfmac: brcmf_vif_add_validate: ... there is already a monitor interface, returning EOPNOTSUPP`

`root@pwnny-cage:~# pwnagotchi --version
1.7.51`