aluminum-ice/pwnagotchi #22: [BUG] Sometimes after rebooting automatically, it boots back up in MANUAL mode

rsun0525, 2023-08-18:
**Describe the bug**
Sometimes after it reboots automatically (50 epochs without visible access points), it starts up in MANUAL mode

**To Reproduce**
Steps to reproduce the behavior:
1. ... Just let run and wait for it to automatically reboot, check localhost:8080 and it says it's in MANUAL, so I ssh in and manually reboot it and it's back in AUTO mode (which then goes to AI mode, etc.)

**Expected behavior**
Whenever it reboots automatically, to restart in AUTO and not MANUAL

**Screenshots**

**Environment (please complete the following information):**
 - Pwnagotchi version 1.7.5
 - OS version  5.10.103-v7+
 - Type of hardware Raspberry Pi 3B+
 - Any additional hardware used -  external USB WiFi adaptor used (while I have disabled the onboard WiFi), I keep the 3B+ plugged into Ethernet.

**Additional context**
Add any other context about the problem here.

rsun0525, 2023-08-22:
I think this bug is related to be using an external USB WIFi adaptor, I re-enabled the onboard WiFi adaptor and haven't had the issue.  Not sure if this can give you anymore information on solving it.

Saru13, 2023-09-24:
run
`touch /root/.pwnagotchi-auto`
This will create that auto file forcing to always boot in auto mode even if the USB or ether cables are connected.

rsun0525, 2023-09-24:
Thanks for the tip!

closed 2023-09-24 by rsun0525