aluminum-ice/pwnagotchi #38: [BUG] mon0 never starts up on a Raspberry Pi 4B built in WiFi adaptor.

rsun0525, 2023-09-04:
**Describe the bug**
I installed 1.7.7 on a Raspberry Pi 4B and mon0 never starts

**To Reproduce**
Steps to reproduce the behavior:
Put 1.7.7 image on MicroSD card, put MicroSD card in Raspberry Pi 4B, power up

**Expected behavior**
I expect it just to start up automatically after powering it on.

**Screenshots**
[image]

**Environment (please complete the following information):**
 - Pwnagotchi version 1.7.7
 - OS version Linux pwnagotchi4b 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
 - Type of hardware  Raspberry Pi 4 Model B Rev 1.1
 - Any additional hardware used None

**Additional context**
I successfully installed 1.7.7 on a Raspberry Pi 400 (disabled onboard Wi-Fi and used an external USB Wi-Fi Adapter), and on a Raspberry Pi Zero 2W, so I think I have my configurations correct, etc.

The only odd thing I noticed, is once I powered it on, I ssh'ed in and did ifconfig -a and saw that not even wlan0 was showing up, but waited a few seconds, I could see wlan0 being there, but still not mon0

peretz27, 2023-09-07:
Can confirm that I have the same issue on a 4B.

brianhealey, 2023-09-13:
I was able to resolve this by manually reinstalling nexmon, and copying the updated `ko` file.  I rebooted afterwards and the monitoring port came up fine.

rsun0525, 2023-09-24:
Closing ticket as said this build is for Raspberry Pi Zero 2W

closed 2023-09-24 by rsun0525