aluminum-ice/pwnagotchi #39: [BUG] Suddenly blind

rsun0525, 2023-09-04:
**Describe the bug**
Pwnagotchi suddenly becomes blind.

Raspberry Pi 400 with external USB WiFi adapter, everything is working great and suddenly it's blind, here's the /var/log/pwnagotchi.log (scroll from bottom up to see).
[pwnagotchi.log](https://github.com/aluminum-ice/pwnagotchi/files/12516452/pwnagotchi.log)

rsun0525, 2023-09-04:
I think I was able to reproduce the issue again on my Raspberry Pi Zero 2 W using the built in WiFi, here's the /var/log/pwnagotchi.log (scroll from the bottom up to see).
[pwnagotchi.log](https://github.com/aluminum-ice/pwnagotchi/files/12516482/pwnagotchi.log)

rsun0525, 2023-09-04:
Here's my 400 again, more times when it died, I just set it to if it's blind for 10 epochs to restart now, not sure if you want me to continue to upload these log files?
[pwnagotchi.log](https://github.com/aluminum-ice/pwnagotchi/files/12516931/pwnagotchi.log)

rsun0525, 2023-09-05:
No worries, I'm assuming the 4B problem is that you need to update some driver, as if I use the external USB Wi-Fi adaptor on it and disable the onboard Wi-Fi adapter it works, etc.

rsun0525, 2023-09-05:
> @rsun0525 Can you please edit "sudo nano /usr/bin/pwnagotchi-launcher" to be in debug mode when you start AUTO mode (add the --debug flag as you see below) and attach the log file for the RPiZ2W when it crashes? Please check the log file to ensure you see [DEBUG] entries.
> 
> ```
> if is_auto_mode; then
>   /usr/local/bin/pwnagotchi --debug
> else
>   /usr/local/bin/pwnagotchi --manual
> fi
> ```

Done, I've got 1.7.7 running on a Raspberry Pi 400 (with external USB WiFi Adaptor) and a Raspberry Pi Zero 2W, both updated and rebooted just now.

rsun0525, 2023-09-05:
rebooted again
[pwnagotchi.log](https://github.com/aluminum-ice/pwnagotchi/files/12524352/pwnagotchi.log)

rsun0525, 2023-09-05:
Sounds good, as soon as I can put 1.7.8, I'll put it on it.  Thanks again for your time and effort!

wpa-2, 2023-09-06:
> > Also if you want me to give you a Raspberry Pi 4B, I've got one I can send you, it's the least I could do given all the time you put in in making your fork. If so, just private message me your address.
> 
> I appreciate it but I have several RPi4B sitting around my desk. I don't need another one at this time. I don't use them for pwnagotchi. Maybe after I feel that the RPiZ2W is working the way I want it to, I will test the 4B and see how well it works on that one.
> 
> I have an alpha v1.7.8 version coming I want you to test to see if this problem is fixed. I will post a note here when its ready.

Thanks for all the awesome work .

rsun0525, 2023-09-11:
Raspberry Pi Zero 2W running 1.7.8 

[pwnagotchi.log](https://github.com/aluminum-ice/pwnagotchi/files/12579704/pwnagotchi.log)

rsun0525, 2023-09-12:
Yes, I've seen crashes which I think are related to websockets.

insanelysick, 2023-09-12:
Took your latest built out for a trip through the city yesterday and started getting the below for tens of thousands of lines
"[2023-09-11 08:01:26,589] [ERROR] Lost websocket connection. Reconnecting..."

For giggles I took your bettercap.py and put it into Xbits build and didn’t get these errors for some reason.

insanelysick, 2023-09-12:
> > Took your latest built out for a trip through the city yesterday and started getting the below for tens of thousands of lines
> > "[2023-09-11 08:01:26,589] [ERROR] Lost websocket connection. Reconnecting..."
> > For giggles I took your bettercap.py and put it into Xbits build and didn’t get these errors for some reason.
> 
> That's the way it's supposed to work. Before you never saw the error because bettercap was simply spinning up more and more websockets instead of reconnecting to an existing socket until it exhausted all the ephemeral TCP ports and then did a hard crash. What you're seeing is the correct behavior and not a crash.

Interesting, my unit became unresponsive and I had to pull the plug then found that error about 40,000 times in the logs. I’ll give it another run and try waiting out any unresponsiveness and see what happens

insanelysick, 2023-09-13:
Thans @aluminum-ice, I've burned a fresh copy of 1.7.8 and replaced bettercap.py with the above. I'll run it for a while and report back although I don't think I'll be heading into an area with a high amount of APs until early next week but I'll see what I can do.

insanelysick, 2023-09-13:
Got these in the logs so far

[2023-09-13 11:05:22,636] [ERROR] Request connection error (HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x462a3bf0>: Failed to establish a new connection: [Errno 111] Connection refused'))) while running command (wifi.recon.channel 1,2,5,7,8,9,11,13)
    raise err
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
[2023-09-13 11:05:23,660] [ERROR] Request connection error (HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x44367630>: Failed to establish a new connection: [Errno 111] Connection refused'))) while running command (wifi.recon.channel 1,2,5,7,8,9,11,13)
    raise err
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
    return decode(r, verbose_errors=verbose_errors)
[2023-09-13 11:05:24,070] [ERROR] Request connection error (HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x44355230>: Failed to establish a new connection: [Errno 111] Connection refused'))) while running command (set wifi.ap.ttl 368)
    raise err
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
[2023-09-13 11:05:24,123] [ERROR] Request connection error (HTTPConnectionPool(host='localhost', port=8081): Max retries exceeded with url: /api/session (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x44355490>: Failed to establish a new connection: [Errno 111] Connection refused'))) while running command (set wifi.ap.ttl 368)
    raise err
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
[2023-09-13 11:05:26,099] [ERROR] [ai] error while training (local variable 'r' referenced before assignment)
    return decode(r, verbose_errors=verbose_errors)

kennyHH, 2023-09-13:
> @insanelysick @rsun0525 Please replace your bettercap.py file with the following code and let me know if you see the same problem "[2023-09-11 08:01:26,589] [ERROR] Lost websocket connection. Reconnecting..."
> 
> ```
> import logging
> import requests
> import websockets
> import asyncio
> import random
> 
> from requests.auth import HTTPBasicAuth
> from time import sleep
> 
> requests.adapters.DEFAULT_RETRIES = 5 # increase retries number
> 
> ping_timeout  = 90
> ping_interval = 60
> 
> max_sleep = 2.0
> 
> def decode(r, verbose_errors=True):
>     try:
>         return r.json()
>     except Exception as e:
>         if r.status_code == 200:
>             logging.error("error while decoding json: error='%s' resp='%s'" % (e, r.text))
>         else:
>             err = "error %d: %s" % (r.status_code, r.text.strip())
>             if verbose_errors:
>                 logging.info(err)
>             raise Exception(err)
>         return r.text
> 
> 
> class Client(object):
>     def __init__(self, hostname='localhost', scheme='http', port=8081, username='user', password='pass'):
>         self.hostname = hostname
>         self.scheme = scheme
>         self.port = port
>         self.username = username
>         self.password = password
>         self.url = "%s://%s:%d/api" % (scheme, hostname, port)
>         self.websocket = "ws://%s:%s@%s:%d/api" % (username, password, hostname, port)
>         self.auth = HTTPBasicAuth(username, password)
> 
>     def session(self):
>         r = requests.get("%s/session" % self.url, auth=self.auth)
>         return decode(r)
> 
>     async def start_websocket(self, consumer):
>         s = "%s/events" % self.websocket
> 
>         # restarted every time the connection fails
>         while True:
>             logging.info("creating new websocket...")
>             try: 
>                 async with websockets.connect(s, ping_interval=ping_interval, ping_timeout=ping_timeout) as ws:
>                     # listener loop
>                     while True:
>                         try:
>                             async for msg in ws:
>                                 try:
>                                     await consumer(msg)
>                                 except Exception as ex:
>                                         logging.debug("error while parsing event (%s)", ex)
>                         except websockets.exceptions.ConnectionClosedError:
>                             try: 
>                                 pong = await ws.ping()
>                                 await asyncio.wait_for(pong, timeout=ping_timeout)
>                                 logger.warning('ping OK, keeping connection alive...')
>                                 continue
>                             except:
>                                 sleep_time = max_sleep*random.random()
>                                 logger.warning('ping error - retrying connection in {} sec'.format(sleep_time))
>                                 await asyncio.sleep(sleep_time)
>                                 break
>             except ConnectionRefusedError:
>                 sleep_time = max_sleep*random.random()
>                 logger.warning('nobody seems to listen to the bettercap endpoint...')
>                 logger.warning('retrying connection in {} sec'.format(sleep_time))
>                 await asyncio.sleep(sleep_time)
>                 continue
> 
>     def run(self, command, verbose_errors=True):
>         for _ in range(0,2):
>             try:
>                 r = requests.post("%s/session" % self.url, auth=self.auth, json={'cmd': command})
>             except requests.exceptions.ConnectionError as e:
>                 sleep_time = max_sleep*random.random()
>                 logging.exception("Request connection error (%s) while running command (%s)", e, command)
>                 logging.warning('Retrying run in {} sec'.format(sleep_time))
>                 sleep(sleep_time)
>             else:
>                 break
> 
>         return decode(r, verbose_errors=verbose_errors)
> ```

I had the same issue as other guys. My last log file has over 4 million lines of "[ERROR] Lost websocket connection. Reconnecting..." in approximately 40 minutes. I'll try your new solution tomorrow, and I've also included zipped log file since it's over 330mb in size when unpacked :)
[pwnagotchi-6.gz](https://github.com/aluminum-ice/pwnagotchi/files/12602485/pwnagotchi-6.gz)

rsun0525, 2023-09-16:
I "think" there's been an improvement since implementing the bettercap.py above, at least when it is blind, it's autorebooting itself.  I'll try to put it on more pwnagotchi's and see if I can create more chances for it to fail and send up some logs.  Thanks again for your time and effort, it's much appreicated!

rsun0525, 2023-09-16:
> Yes I tested it extensively yesterday. The code no longer crashes out or
> lock up. Instead, when the web socket loses the connection it reconnects
> after a few seconds. However, I noticed that this inevitably seems to show
> at that point the device is blind. I suspect it's signaling that the nexmon
> driver has crashed or something. So my pwnagotchi reboots soon after.
>
> I've also noticed that my pwnagotchi is now far better at capturing
> handshakes. At least qualitatively it's more successful at seeing APs and
> getting handshakes.

Next thing to fix is support Raspberry Pi 4b internal wifi adaptor :-).

insanelysick, 2023-09-16:
> Yes I tested it extensively yesterday. The code no longer crashes out or lock up. Instead, when the web socket loses the connection it reconnects after a few seconds. However, I noticed that this [...]

I’ve noticed that the watchdog seems to think it’s blind and does a reboot within 4-10 mins of boot on mine as I can see some brcm errors, however disabling the watchdog stopped this for me. Just got back from a trip to the shops and in got 90 handshakes in 45 mins, 0 blind epochs and it’s still going as we speak!

rsun0525, 2023-09-16:
Thanks for the tip. I will turn off watchdog and see.

insanelysick, 2023-09-16:
Checked in on the logs at the end of the day. 129 handshakes captured, no bettercap web socket issues even when in an area infested with APs and potentially 400-500 clients. Seems like it's sorted but will take it through the city on Monday to make sure. Thanks for the hard work dude, its been a long time coming but I think the end is in sight for getting the 02 stable.

ryanbeard82, 2023-09-16:
Running smooth as eggs on this end too. Been running around all day with it in my pocket. No problemos.

rsun0525, 2023-09-17:
For a crash on my Raspberry Pi 3B+

[image]

ifconfig still shows the device there:

[image]

[pwnagotchi.log](https://github.com/aluminum-ice/pwnagotchi/files/12642959/pwnagotchi.log)

rsun0525, 2023-09-21:
Saw this, and I had to reboot it to start working again.
[image]
[pwnagotchi.log](https://github.com/aluminum-ice/pwnagotchi/files/12678638/pwnagotchi.log)

rsun0525, 2023-09-21:
What I notice is that when my pwnagotchi hit this condition, I know to ssh in and just "reboot" to get it going again.

[image]

Is there some additional code you can add that say if you are in this state for several cycles, just do a "sudo reboot" on it?

rsun0525, 2023-09-22:
In 1.7.9, Sometimes after a reboot (manual or automaticaly), I can't get it to start up, and the only solution I can find it to reboot again and it does start up.  

[image]

[pwnagotchi.log](https://github.com/aluminum-ice/pwnagotchi/files/12699124/pwnagotchi.log)

rsun0525, 2023-09-24:
Going to close this out open up a new ticket specific to bettercap.py failing

closed 2023-09-24 by rsun0525