aluminum-ice/pwnagotchi #54: Full 64-bit os for Rpi Zero2W?

amenekowo, 2023-10-18:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
We just dropped the support of Rpi Zero W in 1.7.7, so I wondered that we could use a arm64 image of raspbian at build. According to Jeff's YouTube video, arm64 seems have a better performance, so maybe it helps pwnagotchi too.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Make built OS fully arm64
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
I think it could be possible because we build nexmon from source. Also, tensorflow could also build for arm64. Just wondered how much time we need to build this.

waterfallm, 2023-11-05:
i might have something soon for this as i'm looking at bumping the version of the OS to get my GeekWorm x306 working

jayofelony, 2024-01-18:
> @jayofelony started a fork for 64-bit Raspberry Pi's. You can find it [here](https://github.com/jayofelony/pwnagotchi-bookworm)

🎉🎉 And boy oh boy does it work well.. 🫡

Opalinus, 2024-01-18:
> > @jayofelony started a fork for 64-bit Raspberry Pi's. You can find it [here](https://github.com/jayofelony/pwnagotchi-bookworm)
> 
> Yes, it's good work. This is my hobby tho and I like working on my fork.
> 
> I'm close to a fully working 64bit stable baseline3 version. Been running it locally with a few remaining small fixes needed.

Can't wait to test it, since my PiSugar2 doesn't seem to work with armv7l. I found this [issue](https://github.com/PiSugar/pisugar-power-manager-rs/issues/53)

Maynardnaze, 2024-02-07:
Are we waiting for Waveshare to release an update?