aluminum-ice/pwnagotchi #61: [BUG] Bettercap

shashankx86, 2023-11-04:
**Describe the bug**
Bettercap not running

Display: false (using webUI)

**Expected behavior**
Run pwnagotchi

**Environment (please complete the following information):**
 - Pwnagotchi: 1.7.9
 - OS version: Linux pwnagotchi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
 - Raspberry pi 4

**Additional context**
Bettercap not running and stuck on "waiting for bettercap API to be available"

```
pi@pwnagotchi:~ $ tail -f /var/log/pwnagotchi.log
[2023-05-03 08:18:14,321] [INFO] waiting for bettercap API to be available ...
[2023-05-03 08:18:15,331] [INFO] waiting for bettercap API to be available ...
[2023-05-03 08:18:16,339] [INFO] waiting for bettercap API to be available ...
[2023-05-03 08:18:17,349] [INFO] waiting for bettercap API to be available ...
[2023-05-03 08:18:18,359] [INFO] waiting for bettercap API to be available ...
...

```
pi@pwnagotchi:~ $ journalctl -fu bettercap
-- Logs begin at Wed 2023-05-03 07:49:02 IST. --
May 03 08:18:47 pwnagotchi systemd[1]: bettercap.service: Scheduled restart job, restart counter is at 56.
May 03 08:18:47 pwnagotchi systemd[1]: Stopped bettercap api.rest service..
May 03 08:18:47 pwnagotchi systemd[1]: Started bettercap api.rest service..
May 03 08:18:47 pwnagotchi bettercap-launcher[3940]: grep: /sys/class/net/mon0/operstate: No such file or directory
May 03 08:18:47 pwnagotchi bettercap-launcher[3940]: command failed: Operation not supported (-95)
May 03 08:18:48 pwnagotchi bettercap-launcher[3940]: WARNING: Terminal colors have been disabled, view will be very limited.
May 03 08:18:48 pwnagotchi bettercap-launcher[3940]: bettercap v2.32.0 (built for linux arm with go1.20.6) [type 'help' for a list of commands]
May 03 08:18:48 pwnagotchi bettercap-launcher[3940]: no interface matching 'mon0' found.
May 03 08:18:48 pwnagotchi systemd[1]: bettercap.service: Main process exited, code=exited, status=1/FAILURE
May 03 08:18:48 pwnagotchi systemd[1]: bettercap.service: Failed with result 'exit-code'.
```

```
pi@pwnagotchi:~ $ sudo ifconfig -a
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether e4:5f:01:0e:b7:79  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 4093  bytes 261299 (255.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 4093  bytes 261299 (255.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.2  netmask 255.255.255.0  broadcast 10.0.0.255
        inet6 fe80::6ce0:feff:fe54:8875  prefixlen 64  scopeid 0x20<link>
        ether 6e:e0:fe:54:88:75  txqueuelen 1000  (Ethernet)
        RX packets 6683  bytes 641701 (626.6 KiB)
        RX errors 0  dropped 10  overruns 0  frame 0
        TX packets 5793  bytes 1834576 (1.7 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether e4:5f:01:0e:b7:7a  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

shashankx86, 2023-11-04:
tried change user and pass for bettercap and reboot
still same

shashankx86, 2023-11-04:
tried changing main.iface "mon0" to "wlan0"
restart, reboot 
still getting error with mon0

itsNeutron, 2023-11-04:
Same here on same hardware.

shashankx86, 2023-11-04:
> Same here on same hardware.

any be rpi 4 specific issue

shashankx86, 2023-11-04:
@aluminum-ice help!!

shashankx86, 2023-11-04:
i searched for wlan0 and there commands to wlan0 
the one commit i found https://github.com/aluminum-ice/pwnagotchi/commit/107af9a734e5eab8efee3ba92e128c7a6d15105a

my guess is wlan0 is for on board wifi and mon0 for external adapter 
guess not sure

@aluminum-ice ???

shashankx86, 2023-11-04:
sorry if it caused any inconvenience
can just tell me what "could" be the reason's 
i am completely clueless what to do and whats causing this error

and
Thank you keeping pwnagotchi updated

shashankx86, 2023-11-04:
and that was time i tagged you, will not form now on

sizzlik, 2023-11-08:
the settings from the config.toml dont get set correctly. Check the user/pass in /etc/pwnagotchi/config.toml and make sure it is the same in /usr/local/share/bettercap/caplets/pwnagotchi-auto.cap and /usr/local/share/bettercap/caplets/pwnagotchi-manual.cap

For some reason setting you make in the config.toml (no matter if direct sd edit, ssh edit or webcfg) are not setup in other files. Took me quete a while to change my pwnagotchi name. It changed on my ink display, but to connect i still needed pwnagotchi.local and not newname.local..but bettercap needed to connect to newname.api...no clue whats causing it..

itsNeutron, 2023-11-08:
If the author lets me I’ve found a repo about a pwnagotchi made for RPi4-3 that works out of the box
Inviato da iPhone

shashankx86, 2023-11-08:
> If the author lets me I’ve found a repo about a pwnagotchi made for RPi4-3 that works out of the box

tell me

tubalainen, 2023-11-10:
Same issue with RPI0W as well. I am trying to get 1.7.9 working "out of the box" on a brand new SD card.
In AUTOMODE
```
tail -f /var/log/pwnagotchi.log
[2023-11-10 09:57:38,654] [ERROR] error connecting to the pwngrid-peer service: HTTPConnectionPool(host='127.0.0.1', port=8666): Max retries exceeded with url: /api/v1/data (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb0b722f0>: Failed to establish a new connection: [Errno 111] Connection refused'))
[2023-11-10 09:57:43,709] [ERROR] error connecting to the pwngrid-peer service: HTTPConnectionPool(host='127.0.0.1', port=8666): Max retries exceeded with url: /api/v1/data (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xadac9890>: Failed to establish a new connection: [Errno 111] Connection refused'))
[2023-11-10 09:57:48,769] [ERROR] error connecting to the pwngrid-peer service: HTTPConnectionPool(host='127.0.0.1', port=8666): Max retries exceeded with url: /api/v1/data (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xadac9330>: Failed to establish a new connection: [Errno 111] Connection refused'))
[2023-11-10 09:57:53,849] [ERROR] error connecting to the pwngrid-peer service: HTTPConnectionPool(host='127.0.0.1', port=8666): Max retries exceeded with url: /api/v1/data (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xadae5830>: Failed to establish a new connection: [Errno 111] Connection refused'))
[2023-11-10 09:57:58,928] [ERROR] error connecting to the pwngrid-peer service: HTTPConnectionPool(host='127.0.0.1', port=8666): Max retries exceeded with url: /api/v1/data (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb0b4dd10>: Failed to establish a new connection: [Errno 111] Connection refused'))
[2023-11-10 09:58:03,996] [ERROR] error connecting to the pwngrid-peer service: HTTPConnectionPool(host='127.0.0.1', port=8666): Max retries exceeded with url: /api/v1/data (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xadac9430>: Failed to establish a new connection: [Errno 111] Connection refused'))
[2023-11-10 09:58:09,050] [ERROR] error connecting to the pwngrid-peer service: HTTPConnectionPool(host='127.0.0.1', port=8666): Max retries exceeded with url: /api/v1/data (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb0b72770>: Failed to establish a new connection: [Errno 111] Connection refused'))
***********
tail -f /var/log/syslog
Nov 10 10:03:52 pwnagotchi systemd[1]: bettercap.service: Service RestartSec=30s expired, scheduling restart.
Nov 10 10:03:52 pwnagotchi systemd[1]: bettercap.service: Scheduled restart job, restart counter is at 28.
Nov 10 10:03:52 pwnagotchi systemd[1]: Stopped bettercap api.rest service..
Nov 10 10:03:52 pwnagotchi systemd[1]: Started bettercap api.rest service..
Nov 10 10:03:52 pwnagotchi bettercap-launcher[3052]: grep: /sys/class/net/mon0/operstate: No such file or directory
Nov 10 10:03:52 pwnagotchi rfkill: unblock set for all
Nov 10 10:03:52 pwnagotchi kernel: [  947.597588] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
Nov 10 10:03:52 pwnagotchi bettercap-launcher[3052]: command failed: Operation not supported (-95)
Nov 10 10:03:53 pwnagotchi bettercap-launcher[3052]: WARNING: Terminal colors have been disabled, view will be very limited.
Nov 10 10:03:53 pwnagotchi bettercap-launcher[3052]: bettercap v2.32.0 (built for linux arm with go1.20.6) [type 'help' for a list of commands]
Nov 10 10:03:53 pwnagotchi bettercap-launcher[3052]: no interface matching 'mon0' found.
Nov 10 10:03:53 pwnagotchi systemd[1]: bettercap.service: Main process exited, code=exited, status=1/FAILURE
Nov 10 10:03:53 pwnagotchi systemd[1]: bettercap.service: Failed with result 'exit-code'.
Nov 10 10:03:57 pwnagotchi systemd[1]: pwngrid-peer.service: Service RestartSec=30s expired, scheduling restart.
Nov 10 10:03:57 pwnagotchi systemd[1]: pwngrid-peer.service: Scheduled restart job, restart counter is at 28.
Nov 10 10:03:57 pwnagotchi systemd[1]: Stopped pwngrid peer service..
Nov 10 10:03:57 pwnagotchi systemd[1]: Started pwngrid peer service..
Nov 10 10:03:59 pwnagotchi systemd[1]: pwngrid-peer.service: Main process exited, code=exited, status=1/FAILURE
Nov 10 10:03:59 pwnagotchi systemd[1]: pwngrid-peer.service: Failed with result 'exit-code'.

```

sizzlik, 2023-11-10:
> Nov 10 10:03:53 pwnagotchi bettercap-launcher[3052]: no interface matching 'mon0' found.

this could be your problem..mon0 isnt your wifi adapter. Check "ip a" or the depricated ifconfig to check the name of you wifi/internet bound adapter. maybe you see something along the line of "wlpxsx" where x is some number

tubalainen, 2023-11-10:
> this could be your problem..mon0 isnt your wifi adapter. Check "ip a" or the depricated ifconfig to check the name of you wifi/internet bound adapter. maybe you see something along the line of [...]

100% agreed. Its an older version of Pi Zero. This for does not have support for that chipset (as of now). No wifi interface is detected.

shashankx86, 2023-11-10:
update:

i was able to get it working with my usb wifi adapter 
(disabled inbuilt wifi)

if anyone knows a way to get wifi (inbuilt) let me know

noncenz, 2023-11-22:
I'm seeing this issue on a Zero 2W. Bettercap comes up fine if I have the USB interface plugged in but not when I'm tethered over bluetooth. 

Further investigation shows Bettercap-launcher throwing the error `Grep /sys/class/net/eth0/operstate No such file or directory` when switching into manual mode. Due to that error, the launcher script mistakenly runs Bettercap with the `auto` caplet instead of the `manual` version. Looks like it's arising from[ this changesst](https://github.com/aluminum-ice/pwnagotchi/commit/b9c7cb34ee4778d5df84dcba72dbccd20864a092), which was fixing a previous bug. 

I'm just getting my feet wet in this codebase, hope this helps.

noncenz, 2023-11-23:
I played around with this a little more and got closer to root cause:

It looks like there is a race condition in __init__.py `restart(mode)` between writing, reading and deleting the auto/manual override flag files. It causes bettercap-launcher to sometimes launch with the auto caplet when it should be the manual one. You can detect the issue with `tail -n 50 /var/log/daemon.log` and observing which interface the Bettercap api comes up on.

This explains why the bug doesn't surface when connected via USB, as presence of the USB interface being up satisfies a different branch in the `is_auto_mode` logic. 

As a hack, in __init__.py function `restart(mode)` you can add `time.sleep(4)` between the two os.system calls that restart the Bettercap and Pwnagotchi services.

My guess here is that the Pwnagotchi service is deleting the flag file before the Bettercap service has a chance to test for it. The os.system calls used here seem to be blocking on this OS but perhaps the error we get earlier in the Bettercap-launcher script is returning control back to python.....

noncenz, 2023-11-24:
@shashankx86 I'll PR the change today as a temporary fix but really want to get at root cause. You should know that the performance of the Bettercap ui over bluetooth tethering isn't great; separate issue that seems to exist all the way back to the original release.