aluminum-ice/pwnagotchi #63: [BUG]

joejsullivan, 2023-11-05:
**Describe the bug**
It appears following the 1.8.0 update, pwnagochi does not start. The log shows an update available, downloaded, then applied. I wouldn't rule out that I'm doing something wrong.

```
[INFO] [update] checking for updates ...
[2023-11-05 08:17:36,815] [WARNING] update for aluminum-ice/pwnagotchi available (local version is '1.7.9'): https://github.com/aluminum-ice/pwnagotchi/archive/1.8.0.zip
[2023-11-05 08:17:36,817] [INFO] [update] downloading https://github.com/aluminum-ice/pwnagotchi/archive/1.8.0.zip to /tmp/updates/pwnagotchi/pwnagotchi_1.8.0.zip ...
[2023-11-05 08:17:39,600] [INFO] [update] extracting /tmp/updates/pwnagotchi/pwnagotchi_1.8.0.zip to /tmp/updates/pwnagotchi ...
[2023-11-05 08:17:40,937] [INFO] [update] installing pwnagotchi ...
[2023-11-05 08:18:13,779] [INFO] [update] done
[2023-11-05 08:18:13,809] [WARNING] rebooting ...
[2023-11-05 08:18:23,850] [WARNING] syncing...
```

**To Reproduce**
Steps to reproduce the behavior:
1. Flash 1.7.9
2. Restart will force the udpate
3. Pwnagotchi will fail to start

**Expected behavior**
Service should start and run

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Pwnagotchi version 1.8.0
 - Raspbian GNU/Linux 10 (buster)
 - Linux pwnagotchi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
 - pi 2 W
 - Waveshare 3
 - piSugar 2

**Additional context**
Output of trying to start service:

```
 pwnagotchi
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 152, in <module>
    plugins.load(config)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 124, in load
    enabled = [name for name, options in config['main']['plugins'].items() if
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 125, in <listcomp>
    'enabled' in options and options['enabled']]
TypeError: argument of type 'int' is not iterable
```

Slayingripper, 2023-11-05:
I am facing the exact same issue, it was working fine all day yesterday with no issues, but after the update was applied it stopped working. I've gone through my Config file , even trying to use the default but the problem still persists

Slayingripper, 2023-11-05:
I've rolled back to the previous version for now and turned off auto update until further notice

DW4y, 2023-11-05:
Have the same issue. I yesterday everything works.

I have set up a new pwnagotchi three times with min. and max. config. All works fine until it updates, after this the error: `ExecStart=/usr/bin/pwnagotchi-launcher (code=exited, status=1/FAILURE)`

Same Setup:

- pi 2 W
- Waveshare 3
- piSugar 2

joejsullivan, 2023-11-05:
Auto update is still on. I re flashed, restarted and got nabbed by the update again.

[2023-11-05 20:29:10,667] [INFO] [update] checking for updates ...
[2023-11-05 20:29:13,496] [WARNING] update for aluminum-ice/pwnagotchi available (local version is '1.7.9'): https://github.com/aluminum-ice/pwnagotchi/archive/1.8.0.zip
[2023-11-05 20:29:13,507] [INFO] [update] downloading https://github.com/aluminum-ice/pwnagotchi/archive/1.8.0.zip to /tmp/updates/pwnagotchi/pwnagotchi_1.8.0.zip ...
[2023-11-05 20:29:16,373] [INFO] [update] extracting /tmp/updates/pwnagotchi/pwnagotchi_1.8.0.zip to /tmp/updates/pwnagotchi ...
[2023-11-05 20:29:17,240] [INFO] [update] installing pwnagotchi ...
[2023-11-05 20:29:50,187] [INFO] [update] done
[2023-11-05 20:29:50,217] [WARNING] rebooting ...
[2023-11-05 20:30:00,257] [WARNING] syncing...

Result post restart:
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 152, in <module>
    plugins.load(config)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 124, in load
    enabled = [name for name, options in config['main']['plugins'].items() if
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 125, in <listcomp>
    'enabled' in options and options['enabled']]
TypeError: argument of type 'int' is not iterable