aluminum-ice/pwnagotchi #72: [BUG] Auto-Update from 1.7.9 > 1.8.0 breaks pwnagotchi.service

MykeHalk, 2023-11-20:
Same as #63 

**Steps to reproduce the behavior:**
1. Install 1.7.9
2. Connect pwnagotchi to network/internet
3. Allow to auto update.
4. After update pwnagotchi reboots and fails to initialize pwnagotchi.service.

**Expected behavior**
Should update from 1.7.9 to 1.8.0 without breaking service.

**Environment**
 - Pwnagotchi version 1.8.0
 - Pi Zero W 2
 - Waveshare 3

**Additional context**
If flashing 1.8.0 directly without updating from 1.7.9, pwnagotchi.service functions and everything seems to be working.

**"sudo pwnagotchi-launcher" output:**
> pwnagotchi
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 152, in <module>
    plugins.load(config)
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 124, in load
    enabled = [name for name, options in config['main']['plugins'].items() if
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/plugins/__init__.py", line 125, in <listcomp>
    'enabled' in options and options['enabled']]
TypeError: argument of type 'int' is not iterable

Safi-Dev, 2023-11-21:
> Okay so i installed 1.7.9
> 
> edited config.toml to include
> 
> > ui.display.enabled = true
> > ui.display.type = "waveshare_3"
> > ui.display.rotation = 180
> > ui.display.color = "black"
> > main.plugins.bt-tether.enabled = true
> > main.plugins.bt-tether.devices.android-phone.enabled = true
> > main.plugins.bt-tether.devices.android-phone.search_order = 1
> > main.plugins.bt-tether.devices.android-phone.mac = "MYMACADDRESS"
> > main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.44"
> > main.plugins.bt-tether.devices.android-phone.netmask = 24
> > main.plugins.bt-tether.devices.android-phone.interval = 1
> > main.plugins.bt-tether.devices.android-phone.scantime = 10
> > main.plugins.bt-tether.devices.android-phone.max_tries = 10
> > main.plugins.bt-tether.devices.android-phone.share_internet = true
> > main.plugins.bt-tether.devices.android-phone.priority = 1
> > main.custom_plugins = "/usr/local/share/pwnagotchi/custom-plugins/"
> 
> Restarted pwnagotchi with "systemctl restart pwnagotchi"
> 
> Enabled logtail
> 
> Paired bluetooth tether
> 
> Auto updated
> 
> rebooting
> 
> > 
> 
> [2023-11-21 03:19:20,127]	[WARNING]	update for aluminum-ice/pwnagotchi available (local version is '1.7.9'): https://github.com/aluminum-ice/pwnagotchi/archive/1.8.0.zip
> [2023-11-21 03:19:20,148]	[INFO]	[update] downloading https://github.com/aluminum-ice/pwnagotchi/archive/1.8.0.zip to /tmp/updates/pwnagotchi/pwnagotchi_1.8.0.zip ...
> [2023-11-21 03:20:25,316]	[INFO]	[update] extracting /tmp/updates/pwnagotchi/pwnagotchi_1.8.0.zip to /tmp/updates/pwnagotchi ...
> [2023-11-21 03:20:28,239]	[INFO]	[update] installing pwnagotchi ...
> [2023-11-21 03:21:21,700]	[INFO]	[update] done
> [2023-11-21 03:21:22,518]	[WARNING]	rebooting ...
> [2023-11-21 03:21:32,946]	[WARNING]	syncing...
> [image]
> 
> [image]
> 
> Anything else you want me to check?

Can confirm that I received the exact same errors after an auto-update from 1.7.9--> 1.8.0, I flashed my sd card to 1.8.0 and it is fully functional.

buave, 2023-11-21:
I have the same problem
This is my config.toml after auto-update:
```
main.name = "pwnagotchi"
main.lang = "en"
main.whitelist = [
]
main.plugins.grid.enabled = true
main.plugins.grid.report = false
main.plugins.grid.exclude = [
]

main.plugins.auto-update.enabled = true
main.plugins.auto-update.install = true
main.plugins.auto-update.interval = 1

main.plugins.net-pos.enabled = true
main.plugins.net-pos.api_key = "test"

main.plugins.gps.enabled = false
main.plugins.gps.speed = 19200
main.plugins.gps.device = "/dev/ttyUSB0"

main.plugins.webgpsmap.enabled = true

main.plugins.onlinehashcrack.enabled = false
main.plugins.onlinehashcrack.email = "nope"
main.plugins.onlinehashcrack.dashboard = "nope"
main.plugins.onlinehashcrack.single_files = false
main.plugins.onlinehashcrack.whitelist = [
]

main.plugins.wpa-sec.enabled = true
main.plugins.wpa-sec.api_key = "nope"
main.plugins.wpa-sec.api_url = "https://wpa-sec.stanev.org"
main.plugins.wpa-sec.download_results = true
main.plugins.wpa-sec.whitelist = []
main.plugins.wpa-sec.founds.insert.enabled = true

main.plugins.wigle.enabled = false
main.plugins.wigle.api_key = ""
main.plugins.wigle.whitelist = []
main.plugins.wigle.donate = true

main.plugins.bt-tether.enabled = true
main.plugins.bt-tether.devices.android-phone.enabled = true
main.plugins.bt-tether.devices.android-phone.search_order = 1
main.plugins.bt-tether.devices.android-phone.mac = "nope"
main.plugins.bt-tether.devices.android-phone.ip = "192.168.44.100"
main.plugins.bt-tether.devices.android-phone.netmask = 24
main.plugins.bt-tether.devices.android-phone.interval = 1
main.plugins.bt-tether.devices.android-phone.scantime = 10
main.plugins.bt-tether.devices.android-phone.max_tries = 10
main.plugins.bt-tether.devices.android-phone.share_internet = true
main.plugins.bt-tether.devices.android-phone.priority = 1

main.plugins.bt-tether.devices.ios-phone.enabled = false
main.plugins.bt-tether.devices.ios-phone.search_order = 2
main.plugins.bt-tether.devices.ios-phone.mac = ""
main.plugins.bt-tether.devices.ios-phone.ip = "172.20.10.6"
main.plugins.bt-tether.devices.ios-phone.netmask = 24
main.plugins.bt-tether.devices.ios-phone.interval = 5
main.plugins.bt-tether.devices.ios-phone.scantime = 20
main.plugins.bt-tether.devices.ios-phone.max_tries = 0
main.plugins.bt-tether.devices.ios-phone.share_internet = false
main.plugins.bt-tether.devices.ios-phone.priority = 999

main.plugins.memtemp.enabled = true
main.plugins.memtemp.scale = "celsius"
main.plugins.memtemp.orientation = "horizontal"
[... 173 lines cut ...]
personality.bored_num_epochs = 5
personality.sad_num_epochs = 5
personality.bond_encounters_factor = 20000

bettercap.scheme = "http"
bettercap.hostname = "localhost"
bettercap.port = 8081
bettercap.username = "pwnagotchi"
bettercap.password = "nope"
bettercap.handshakes = "/root/handshakes"
bettercap.silence = [
 "ble.device.new",
 "ble.device.lost",
 "ble.device.disconnected",
 "ble.device.connected",
 "ble.device.service.discovered",
 "ble.device.characteristic.discovered",
 "wifi.client.new",
 "wifi.client.lost",
 "wifi.client.probe",
 "wifi.ap.new",
 "wifi.ap.lost",
 "mod.started",
]
```

diogo-abp-costa, 2023-11-22:
> You mean this line? https://github.com/aluminum-ice/pwnagotchi/blob/5ff97c82b73a98a691c06c497ff8dabc505ecca3/pwnagotchi/defaults.toml#L52C1-L52C46 @diogo-abp-costa

Yes, I added that to my config.toml but with quotation marks in the value and the issue was gone.

joelnb, 2023-11-23:
In my case I was able to fix it like this:

```bash
sed -i 's/main.plugins.wpa-sec-download_interval/main.plugins.wpa-sec.download_interval/' /usr/local/lib/python3.7/dist-packages/pwnagotchi/defaults.toml
```

This was because in that `defaults.toml` file it had a `-` where there should have been a `.`.

I don't understand how the upgrade plays into this because I'm not familiar with the update code but my assumption is that `defaults.toml` was not updated by the upgrade - I can see this was fixed in git in https://github.com/aluminum-ice/pwnagotchi/commit/da12b10c0a80e4d0d20ee3e4c094ae6ca497a0d6

joelnb, 2023-11-25:
In my case it was never in `config.toml` but was in `/usr/local/lib/python3.7/dist-packages/pwnagotchi/defaults.toml` and therefore copied to `/etc/pwnagotchi/default.toml` on each restart. As I had never edited that file before I assumed that the change where @retiolus fixed this issue was not being correctly applied by the upgrade process (despite the file being correct in this git repo).

I checked and after changing the `-` to `.` I have these differences compared to the git version:

```diff
 pwnagotchi/defaults.toml | 28 ----------------------------
 1 file changed, 28 deletions(-)

diff --git a/pwnagotchi/defaults.toml b/pwnagotchi/defaults.toml
index a03d099..47799b8 100644
--- a/pwnagotchi/defaults.toml
+++ b/pwnagotchi/defaults.toml
@@ -125,34 +125,6 @@ main.plugins.logtail.max-lines = 10000
 main.plugins.session-stats.enabled = true
 main.plugins.session-stats.save_directory = "/var/tmp/pwnagotchi/sessions/"

-main.plugins.mastodon.enabled = false
-main.plugins.mastodon.instance_url = ""
-main.plugins.mastodon.token = ""
-main.plugins.mastodon.visibility = "unlisted"
-
-main.plugins.ntfy.enabled = false
-main.plugins.ntfy.name = "pwnagotchi"
-main.plugins.ntfy.serverlink = ""
-main.plugins.ntfy.token = ""
-main.plugins.ntfy.priority = "3"
-main.plugins.ntfy.icon = "https://files.catbox.moe/1toze0.jpg"
-main.plugins.ntfy.on_internet_available_enabled = true
-main.plugins.ntfy.on_internet_available_title = ""
-main.plugins.ntfy.on_internet_available_msg = ""
-main.plugins.ntfy.on_internet_available_tags = ""
-main.plugins.ntfy.on_handshake_enabled = true
-main.plugins.ntfy.on_handshake_title = ""
-main.plugins.ntfy.on_handshake_msg = ""
-main.plugins.ntfy.on_handshake_tags = ""
-main.plugins.ntfy.on_peer_detected_enabled = true
-main.plugins.ntfy.on_peer_detected_title = ""
-main.plugins.ntfy.on_peer_detected_msg = ""
-main.plugins.ntfy.on_peer_detected_tags = ""
-main.plugins.ntfy.on_peer_lost_enabled = true
-main.plugins.ntfy.on_peer_lost_title = ""
-main.plugins.ntfy.on_peer_lost_msg = ""
-main.plugins.ntfy.on_peer_lost_tags = ""
-
 main.log.path = "/var/log/pwnagotchi.log"
 main.log.rotation.enabled = true
 main.log.rotation.size = "10M"
```

So it seems I was missing every change made to that file since the 1.7.9 release. Therefore I think I was correct that the auto-updater has not updated that file at all (at least in my case).

joelnb, 2023-11-25:
I had a better look and if you extract [the zip](https://github.com/aluminum-ice/pwnagotchi/archive/refs/tags/1.8.0.zip) for the 1.8.0 release (which the updater uses) you will see it doesn't contain the fix @retiolus made.

You can also do this in the git repo:

```bash
git checkout 1.8.0
grep -r wpa-sec-download_interval
```

And see the fix is not there in the 1.8.0 tag.

kiraitachi, 2023-11-28:
> it a PR? I will merge it in quickly and release v1.8.1 that will hopefully resolve the breakage that happens by the auto-update.

I understand we will need to reflash manually if we are running onto this issue to get the 1.8.1version.

Thanks!