aluminum-ice/pwnagotchi #81: installation instructions (I could be blind)

DeForestDiamond, 2023-12-02:
Hello! 

I can't seem to get this little guy to come alive. I've looked at a few different install instructions and there are quite a few variations out there. Is there instruction specific to this branch any where? My google-fu has failed me or like I said in the title, "I could be blind."

Thanks a million and thank you for making this branch for the rpi zero 2!
Mike

k-dannemand, 2023-12-04:
Get the last release from: https://github.com/aluminum-ice/pwnagotchi/releases
Flash it to your SD card with something like balenaEtcher. 
if you are running on supporet hardware then just put in the SD card an boot it up.

DeForestDiamond, 2023-12-04:
It really is supposed to be that simple, eh? I wonder if my screen is defective then. I read some where that it wont work unless you have a RTC installed--which i don't yet. Would that cause it not to boot? For now I'll re-flash the card and try again.

DeForestDiamond, 2023-12-05:
Waveshare 2.13 v3. 

I fired up the rpi with the stock OD and it worked, thankfully. Waveshare has some test code for the screen so I'm going to try that next.

DeForestDiamond, 2023-12-06:
I'm 99% sure my e-ink is defective...

closed 2023-12-06 by DeForestDiamond