aluminum-ice/pwnagotchi #82: [BUG][ERROR]  error 400: wifi.interface not set or not found

k-dannemand, 2023-12-04:
it see this error when it has been running for long periods of time.
mostly seen when it has been left on overnight and then it gets caught in a sleep state

it could look as if the NIC is in an sleep state or something

```
[2023-12-03 16:02:56,935] [INFO] sleeping 60 seconds then advancing to next epoch to allow for cleanup code to trigger                                                                                                                                                                                                                                                                                                                       
[2023-12-03 16:02:57,291] [INFO] [ai] saving /root/brain.json                                                                                                                                                                                                                                                                                                                                                                                
[2023-12-03 16:02:57,435] [INFO] [ai] setting new policy:                                                                                                                                                                                                                                                                                                                                                                                    
[2023-12-03 16:03:57,109] [INFO] [epoch 633] duration=00:01:00 slept_for=00:00:00 blind=4 sad=246 bored=0 inactive=250 active=0 peers=0 tot_bond=0.00 avg_bond=0.00 hops=0 missed=0 deauths=0 assocs=0 handshakes=0 cpu=0% mem=60% temperature=40C reward=-0.1583596214511041                                                                                                                                                                
[2023-12-03 16:03:57,112] [INFO] [ai] --- training epoch 46/50 ---                                                                                                                                                                                                                                                                                                                                                                           
[2023-12-03 16:03:57,112] [WARNING] 250 epochs with no activity -> angry                                                                                                                                                                                                                                                                                                                                                                     
[2023-12-03 16:03:57,113] [INFO] [ai] REWARD: -0.158360                                                                                                                                                                                                                                                                                                                                                                                      
[2023-12-03 16:03:57,118] [INFO] [ai] observation:                                                                                                                                                                                                                                                                                                                                                                                           
[2023-12-03 16:03:57,124] [INFO]     aps                                                                                                                                                                                                                                                                                                                                                                                                     
[2023-12-03 16:03:57,125] [INFO]     sta                                                                                                                                                                                                                                                                                                                                                                                                     
[2023-12-03 16:03:57,126] [INFO]     peers                                                                                                                                                                                                                                                                                                                                                                                                   
[2023-12-03 16:03:57,132] [INFO] [ai] saving model to /root/brain.nn ...                                                                                                                                                                                                                                                                                                                                                                     
[2023-12-03 16:03:57,347] [INFO] error 400: wifi.interface not set or not found                                                                                                                                                                                                                                                                                                                                                              
[2023-12-03 16:03:57,353] [ERROR] main loop exception due to unavailable wifi device, likely programmatically disabled (error 400: wifi.interface not set or not found)                                                                                                                                                                                                                                                                      
Traceback (most recent call last):                                                                                                                                                                                                                                                                                                                                                                                                           
  File "/usr/local/lib/python3.7/dist-packages/requests/models.py", line 971, in json                                                                                                                                                                                                                                                                                                                                                        
    return complexjson.loads(self.text, **kwargs)                                                                                                                                                                                                                                                                                                                                                                                            
  File "/usr/lib/python3.7/json/__init__.py", line 348, in loads                                                                                                                                                                                                                                                                                                                                                                             
    return _default_decoder.decode(s)                                                                                                                                                                                                                                                                                                                                                                                                        
[... cut ...]
[2023-12-03 16:03:57,479] [INFO] sleeping 60 seconds then advancing to next epoch to allow for cleanup code to trigger                                                                                                                                                                                                                                                                                                                       
[2023-12-03 16:03:57,649] [INFO] [ai] saving /root/brain.json  
```

 - Pwnagotchi version: v1.8.1
 - OS version: 5.10.103-v7+ - Raspbian GNU/Linux 10 (buster)
 - Type of hardware: rpi02w
 - Any additional hardware used: Waveshare V3

k-dannemand, 2023-12-05:
@aluminum-ice i have not made any custom change to your image. so if it do it as default yes. els no.

jjeon-da, 2023-12-17:
trying to follow guerilla setup guide [https://github.com/Xyl0se/Pwnagotchi-new-guerilla-guide] for this fork to help troubleshoot. 

i've changed all the networking settings on the pi to reflect what's in the guide, but am still having difficulty getting a ping to google :-(

 in the guerilla guide, i get to step '2.4 Internet connection sharing for win10/11'  the  scripts folder </usr/local/src/pwnagotchi/scripts>

but there's the tar file from a release? 

'pi@raspberrypi:/usr/local/src/pwnagotchi $ ls
pwnagotchi-1.8.3.tar.gz'

just flagging this as a change for the repo...  i can copy-over the scripts folder into the pi,  but if it should have unpacked automatically,  then soemthing's off..