aluminum-ice/pwnagotchi #86: [BUG]

JalxP, 2023-12-13:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Allow auto-update to run.
2. Try to connect to web-ui

**Expected behavior**
web-ui should be available.

**Screenshots**
N/A

**Environment (please complete the following information):**
 - Pwnagotchi version — v1.8.2
 - OS version — Debian 12
 - Type of hardware — Raspberry Zero 2 w

**Additional context**
After having my Pwnagotchi automatically updated to v.1.8.2 I noticed I could no longer access my bt-tether/web ui, so I checked the output of `systemctl status pwnagotchi `, which gives me this:

> ExecStart=/usr/bin/pwnagotchi-launcher (code=exited, status=1/FAILURE)

I then launched the launcher manually with `sudo /usr/bin/pwnagotchi-launcher`, where I got this error:

```
Traceback (most recent call last):
  File "/usr/local/bin/pwnagotchi", line 9, in <module>
    import pwnagotchi
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/__init__.py", line 8, in <module>
    from pwnagotchi._version import __version__
  File "/usr/local/lib/python3.7/dist-packages/pwnagotchi/_version.py", line 3, in <module>
    __version__ = os.environ["PWN_RELEASE"]
  File "/usr/lib/python3.7/os.py", line 678, in __getitem__
    raise KeyError(key) from None
KeyError: 'PWN_RELEASE'
```

Not sure where `PWN_RELEASE` environment variable should be defined, but `echo $PWN_RELEASE` was empty.

JalxP, 2023-12-13:
Same as https://github.com/aluminum-ice/pwnagotchi/issues/85

JalxP, 2023-12-13:
Check https://github.com/aluminum-ice/pwnagotchi/commit/1db620dac63c6fcf7bacb1c9adcb4417919ef810 & https://github.com/aluminum-ice/pwnagotchi/commit/05fef287b33d5f77a521841080c04a509051067d
We never actually define the `os.environ["PWN_VERSION"] = 1.8.2` anywhere. I also noticed that the regex only worked because it was actually finding a match in the commented line, not sure if this was expected or not.

JalxP, 2023-12-13:
It's a pleasure to help in the project, I already noticed I said some wrong this tho since I got confused both the `PWN_VERSION` & `PWN_RELEASE` variables with each other. The [Makefile change](https://github.com/aluminum-ice/pwnagotchi/commit/4c04db62195fb29fd385fb3ba9837b642635c6d7) is also broken because of [the removal of the comment in the _version.py file](https://github.com/aluminum-ice/pwnagotchi/commit/05fef287b33d5f77a521841080c04a509051067d) it seems.

Either way glad to help!

bicurico, 2023-12-13:
Hi,

Just went through the same issue... Took a while to land here and another while to understand how to fix it.

For those as dumb as I, here is the step by step instruction:

1) ssh pi@10.0.0.2 -> password is "raspberry"
2) cd /usr/local/lib/python3.7/dist-packages/pwnagotchi
3) sudo nano _version.p
4) edit the __version__ line to be: __version__ = "1.8.2"
5) CTRL+X
6) YES
7) ENTER --> file is now saved
8) sudo reboot now
9) wait 1 minute or so
10) ssh pi@10.0.0.2, enter password
11) sudo reboot now
12) wait 1 minute or so
--> open browser: 10.0.0.2:8080

Enjoy!

Thanks a lot for this fork!

hevnsnt, 2023-12-14:
Just in case someone sees this, there are a few mistakes in [bicurico](https://github.com/bicurico)'s instructions above.  I have fixed them down here (all credit to him though, that fixed mine)

- ssh pi@10.0.0.2 -> password is "raspberry"
- cd /usr/local/lib/python3.7/dist-packages/pwnagotchi
- sudo nano _version.py
- edit the version line to be: `__version__ = "1.8.2"`
- CTRL+X
- YES
- ENTER --> file is now saved
- sudo reboot now
- wait 1 minute or so
- ssh pi@10.0.0.2, enter password
- sudo reboot now
- wait 1 minute or so
    --> open browser: 10.0.0.2:8080