aluminum-ice/pwnagotchi #91: [BUG] Hitting `shutdown`, `reboot` or `restart in auto mode` gives me an error

alichaudry, 2023-12-20:
**Describe the bug**
Hitting `shutdown`, `reboot` or `restart in auto mode` gives me an error on the device:8080 UI.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to device:8080
2. Hit any of `Shutdown`, `Reboot`, or `Restart in AUTO mode`, and hit "okay" on the popup
3. Error pops up, saying: `*Bad Request* The CSRF session token is missing.`

**Expected behavior**
The device should shutdown or restart

**Environment (please complete the following information):**
 - Pwnagotchi version: `pwnver`: 1.8.3
 - OS version: `uname -a`: Raspbian GNU/Linux 10 (buster)
 - Type of hardware: Raspberry Pi 3B
 - Any additional hardware used

hevnsnt, 2023-12-29:
It seems like you are using an old / stale page. You either need to clear your cache or force a page refresh before you hit those buttons. This isn't a bug, I would suggest this bug be closed.

alichaudry, 2023-12-29:
I did reload the page when I tried this. I'll try clear the cache and forcing a refresh to see if that fixes it. Are you able to confirm that the latest version is working correctly for you?

hevnsnt, 2024-01-01:
yes, this happens occasionally for me too, but only if I am using a stale page, which is to be expected. The CSRF token is generated per session, it is a security feature of the webpage. This is not a bug, it is operating as expected. I think you just need to clear your cache and/or open the page with a new browser tab.

alichaudry, 2024-01-02:
Yes, tried a different machine + browser and works as expected. Thank you, closing this issue.

closed 2024-01-02 by alichaudry