aluminum-ice/pwnagotchi #102: [BUG] Auto Mode won't start, stuck at "starting monitor interface"

triple-threat-dan, 2024-01-11:
**Describe the bug**
Pwnagotchi won't start in Auto mode. I can ssh and connect in Manual mode. I cannot reach the web portal through `http://10.0.0.2:8080`, but again doing ssh into 10.0.0.2 works just fine. 

Checking the logs by running `tail -f /var/log/pwnagotchi.log` I get this:

```
[2024-01-11 20:55:28,954] [INFO] initializing waveshare v4 display
[2024-01-11 20:55:28,964] [INFO] BT-TETHER: Successfully loaded ...
[2024-01-11 20:55:33,405] [INFO] triplethreatpwny@9f5c320185d8435c2921d37baa98be63e12b61be2759f289ee5716c92950d485 (v1.8.4)
[2024-01-11 20:55:33,407] [INFO] entering auto mode ...
[2024-01-11 20:55:33,417] [INFO] ai disabled
[2024-01-11 20:55:33,497] [INFO] web ui available at http://[::]:8080/
[2024-01-11 20:55:33,537] [INFO] connecting to http://localhost:8081/api ...
[2024-01-11 20:55:33,804] [INFO] starting monitor interface ...
[2024-01-11 20:55:34,565] [INFO] starting monitor interface ...
[2024-01-11 20:55:35,297] [INFO] starting monitor interface ...
[2024-01-11 20:55:36,016] [INFO] starting monitor interface ...
[2024-01-11 20:55:36,793] [INFO] starting monitor interface ...
[2024-01-11 20:55:37,546] [INFO] starting monitor interface ...
[2024-01-11 20:55:38,271] [INFO] starting monitor interface ...
[2024-01-11 20:55:39,044] [INFO] starting monitor interface ...
[2024-01-11 20:55:39,787] [INFO] starting monitor interface ...
[2024-01-11 20:55:40,555] [INFO] starting monitor interface ...
[2024-01-11 20:55:41,277] [INFO] starting monitor interface ...
```

And it goes on forever stuck at that step.

Have tried restarting the pwnagotchi service by running `sudo touch /root/.pwnagotchi-auto && systemctl restart pwnagotchi` and have tried rebooting, both through commandline and unplugging.

`systemctl status pwnagotchi` isn't very helpful:
```
 pwnagotchi.service - pwnagotchi Deep Reinforcement Learning instrumenting bettercap for WiFI pwning.
   Loaded: loaded (/etc/systemd/system/pwnagotchi.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2024-01-11 20:51:14 GMT; 46s ago
     Docs: https://pwnagotchi.ai
 Main PID: 24287 (bash)
    Tasks: 17
   CGroup: /system.slice/pwnagotchi.service
           ├─24287 bash /usr/bin/pwnagotchi-launcher
           └─24291 /usr/bin/python3 /usr/local/bin/pwnagotchi
```

**Environment (please complete the following information):**
 - Pwnagotchi version: v1.8.4
 - Type of hardware: Raspi 0 2w
 - Any additional hardware used: Waveshare v4

triple-threat-dan, 2024-01-11:
Any help is much appreciated! I can paste my config.toml if needed.

lucidyan, 2024-01-12:
@triple-threat-dan check bettercap logs with `sudo journalctl --unit=bettercap.service`, it seems that problems with it (maybe it could not initialize monitor mode or something).

triple-threat-dan, 2024-01-12:
Thanks for the response, here's what I get in the bettercap logs:
```
Jan 12 14:14:00 triplethreatpwny rfkill[30067]: unblock set for all
Jan 12 14:14:01 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:01 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:02 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:03 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:04 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:04 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:05 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:06 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:07 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:07 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:08 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:08 triplethreatpwny rfkill[30210]: unblock set for all
Jan 12 14:14:09 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:10 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:10 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:10 triplethreatpwny rfkill[30253]: unblock set for all
Jan 12 14:14:11 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:11 triplethreatpwny rfkill[30266]: unblock set for all
Jan 12 14:14:12 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:12 triplethreatpwny rfkill[30279]: unblock set for all
Jan 12 14:14:12 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:13 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
Jan 12 14:14:13 triplethreatpwny rfkill[30309]: unblock set for all
Jan 12 14:14:14 triplethreatpwny bettercap-launcher[460]: command failed: Operation not supported (-95)
```

triple-threat-dan, 2024-01-12:
Adding content from /var/log/syslog too in case it helps:

```
Jan 12 19:58:08 triplethreatpwny rfkill: unblock set for all
Jan 12 19:58:08 triplethreatpwny kernel: [19956.059693] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
Jan 12 19:58:08 triplethreatpwny kernel: [19956.059708] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
Jan 12 19:58:08 triplethreatpwny kernel: [19956.096662] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
Jan 12 19:58:08 triplethreatpwny kernel: [19956.096679] brcmfmac: brcmf_vif_add_validate: ... there is already a monitor interface, returning EOPNOTSUPP
Jan 12 19:58:08 triplethreatpwny kernel: [19956.096693] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-95
Jan 12 19:58:09 triplethreatpwny bettercap-launcher[818]: command failed: Operation not supported (-95)
Jan 12 19:58:09 triplethreatpwny rfkill: unblock set for all
Jan 12 19:58:09 triplethreatpwny kernel: [19956.819832] brcmfmac: brcmf_cfg80211_set_power_mgmt: Forcing power management
Jan 12 19:58:09 triplethreatpwny kernel: [19956.819851] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
Jan 12 19:58:09 triplethreatpwny kernel: [19956.862791] brcmfmac: brcmf_vif_add_validate: Attempt to add a MONITOR interface...
Jan 12 19:58:09 triplethreatpwny kernel: [19956.862810] brcmfmac: brcmf_vif_add_validate: ... there is already a monitor interface, returning EOPNOTSUPP
Jan 12 19:58:09 triplethreatpwny kernel: [19956.862827] ieee80211 phy0: brcmf_cfg80211_add_iface: iface validation failed: err=-95
Jan 12 19:58:09 triplethreatpwny bettercap-launcher[818]: command failed: Operation not supported (-95)
```

triple-threat-dan, 2024-01-12:
@lucidyan I ended up just reflashing it with a fresh config and it all works now. Bettercap got borked somehow, that's all I could tell.

Thank you for your help though!

Closing bug.

closed 2024-01-12 by triple-threat-dan

lucidyan, 2024-01-13:
@triple-threat-dan Sorry for the late response, but I want to say that I see this error from time to time, rebooting helps. It seems that this is due to some kind of implicit case when creating/recreating the monitor interface. Glad to see that I helped you a bit!