aluminum-ice/pwnagotchi #104: No default controller available

slabua, 2024-01-13:
**Describe the bug**
Seems like the wifi/bt hardware is not being recognised.

**To Reproduce**
Steps to reproduce the behavior:
1. ssh into pwnagotchi
2. sudo bluetoothctl
3. scan on
4. "No default controller available"

**Expected behavior**
Be able to establish a bluetooth connection
Be able to collect handshakes

**Environment (please complete the following information):**
 - Pwnagotchi version 1.8.4
 - OS version whichever comes with the 1.8.4 image
 - Type of hardware Pi Zero W 2

sylwesterr, 2024-01-14:
Hi,
I just can confirm the issue with running on image 1.8.4.
The hciconfig shows the device being down:
root@pwnagotchi:/# hciconfig 
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:B9:E4:7A  ACL MTU: 1021:8  SCO MTU: 64:1
	DOWN 
	RX bytes:1250 acl:0 sco:0 events:66 errors:0
	TX bytes:810 acl:0 sco:0 commands:66 errors:0

and 'hcitool dev' shows no device:

root@pwnagotchi:/# hcitool dev
Devices:

inserting card with the exact same configuration but on image 1.8.3 - the bluetooth device is detected and seem to be working fine.

slabua, 2024-01-15:
@sylwesterr thanks later I will try the .3

sylwesterr, 2024-01-15:
Good morning,
yesterday evening I was playing with pwnagotchi on 1.8.3, as I wrote, the bluetooth was working fine then, I used internet connection sharing via usb, and at some point I did "sudo apt-get upgrade" and today I noticed that... my bluetooth stopped working - exactly the same issue: in bluetoothctl "No default controller available".
I reviewed the installed updates and I can see there was bluez update installed:

Start-Date: 2024-01-14  19:04:48
Commandline: apt-get upgrade
Requested-By: pi (1000)
Upgrade: **bluez:armhf (5.50-1.2~deb10u3+rpt1, 5.50-1.2~deb10u4)**, libcurl4:armhf (7.64.0-4+deb10u7, 7.64.0-4+deb10u8), openssh-sftp-server:armhf (1:7.9p1-10+deb10u3, 1:7.9p1-10+deb10u4), ssh:armhf (1:7.9p1-10+deb10u3, 1:7.9p1-10+deb10u4), openssh-server:armhf (1:7.9p1-10+deb10u3, 1:7.9p1-10+deb10u4), openssh-client:armhf (1:7.9p1-10+deb10u3, 1:7.9p1-10+deb10u4), ansible:armhf (2.7.7+dfsg-1+deb10u1, 2.7.7+dfsg-1+deb10u2), curl:armhf (7.64.0-4+deb10u7, 7.64.0-4+deb10u8), libcurl3-gnutls:armhf (7.64.0-4+deb10u7, 7.64.0-4+deb10u8)
End-Date: 2024-01-14  19:05:48

I think it's worh checking this track...

farzadb, 2024-01-22:
I'm having the same issue. I've tried to roll back the bluez version to see if I can pinpoint it to that update breaking bt but I don't seem to have the Linux skills to do that.

voodoo1983, 2024-02-11:
I'm also had same issue. I downgraded to version 5.50-1.2~deb10u3+rpt1.
sudo apt install bluez=5.50-1.2~deb10u3+rpt1
sudo apt-mark hold bluez

After reboot everything works fine.

farzadb, 2024-02-12:
> `sudo apt install bluez=5.50-1.2~deb10u3+rpt1`
> `sudo apt-mark hold bluez`

Thanks for posting this. It worked for me when I rolled back bluez. I didn't know the command.

scapegrace13, 2024-03-15:
> I'm also had same issue. I downgraded to version deb10u3 `sudo apt install bluez=5.50-1.2~deb10u3+rpt1` `sudo apt-mark hold bluez`
> 
> After reboot everything works fine.

Thanks this helped me a lot. If anyone needs a hint, here is exact what I have done to fix it.

`
on your host do:
wget http://archive.raspberrypi.org/debian/pool/main/b/bluez/bluez_5.50-1.2~deb10u3+rpt1_armhf.deb 
upload to gotchi with scp
scp ./bluez_5.50-1.2~deb10u3+rpt1_armhf.deb  pi@pwnagotchi:/home/pi

ssh pi@pwnagotchi
sudo -i
dpkg -i /home/pi/bluez_5.50-1.2~deb10u3+rpt1_armhf.deb

<reboot>
`

lokanidao9991, 2024-06-10:
@scapegrace13 This worked flawlessly, thanks for your manual!