aluminum-ice/pwnagotchi #109: [BUG] default user `pi` and password `raspbery` not working?

formula-spectre, 2024-01-18:
**Describe the bug**
I freshly installed the latest pwnagotchi release 1.8.4, but I can't ssh into it; `ssh pi@169.254.117.14` with password `raspberry` does not work

**To Reproduce**
Steps to reproduce the behavior:
1.get 1.8.4 
2. flash it onto a sd card
3. connect the rpi0w2 to the pc 
4. try to login with pi with the ip in the `usb0` interface with user `pi` and password `raspberry`

**Expected behavior**
the espected behaviour is that ssh pi@whatever-ip then inserting `raspberry` as a password should let me in

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.8.4
 - OS version: ?
 - Type of hardware: rasperry pi 0 w version 2
 - waveshare e-ink display v2

**Additional context**
I followed the guide in pwnagotchi.ai, and there it says that the default password is raspberry; I tried to "accidentally" remove the password hash in `/etc/shadow ` and the `x` in `/etc/passwd` in the rootfs of pwnagotchi, but still nothing.. the error is `pi@169.254.20.168: Permission denied (publickey,keyboard-interactive).`

formula-spectre, 2024-01-18:
forgot to mention: I tried to chroot into the rootfs, but then I remembered that i was trying to chroot into an arm system on and amd64 system.. lol

formula-spectre, 2024-01-18:
another thing: maybe it's because I haven't logged in yet, but the waveshare e-ink display displays (pun not intented) nothing, no face, no status; is it normal?

Opalinus, 2024-01-18:
> ssh pi@169.254.117.14

I assume you have connected pwnagotcha to a windows machine, because 169.254.117.14 indicates an invalid IP adress. This vieo might help you; I already timestamped the important bit https://youtu.be/HWfO2Xh5Wbc&t=414

formula-spectre, 2024-01-18:
I actually have plugged my pwnagotchi to my linux host; I skipped the whole create the 10.0.0.1 ip part though

formula-spectre, 2024-01-18:
also, as I said; my eink display is not showing anything; so idk if i'm in manual mode or not (as the video says); I just soldered some gpio headers and slapped the screen on them; did I miss a step?

Opalinus, 2024-01-18:
> I skipped the whole create the 10.0.0.1 ip part though

It will not work when you skip steps, they are vital.

> did I miss a step?

yes. Try the video or the manual. I link it here:

> Wait before removing the SD card as you will need to create one last file on it with the [initial configuration](https://pwnagotchi.ai/configuration/).

formula-spectre, 2024-01-18:
i created the inital configuration, but i skipped the whole IP part because it talked of networkmanager and I don't use it; do i just `ip addr add 10.0.0.1/255.255.255.0 dev usb0`? or do I have to take other steps?

Opalinus, 2024-01-18:
Your network device name should start with "en". Post your dmesg output to debug your problem.

A photo from your e-Ink screen might help to identify it. My model does have a silver sticker labeled with G stuck on the board and I adressed it in issue #105 and added it here #107

formula-spectre, 2024-01-19:
attached: photos of my e-ink display

[image]

[image]

[image]

formula-spectre, 2024-01-19:
> Your network device name should start with "en". Post your dmesg output to debug your problem.  

```
doas dmesg -W
[   89.299006] logitech-hidpp-device 0003:046D:4096.0009: HID++ 4.5 device connected.
[  114.430881] usb 2-1.1: new high-speed USB device number 4 using ehci-pci
[  114.518149] usb 2-1.1: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 5.10
[  114.518159] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  114.518163] usb 2-1.1: Product: RNDIS/Ethernet Gadget
[  114.518166] usb 2-1.1: Manufacturer: Linux 5.10.103-v7+ with 3f980000.usb
[  114.566557] cdc_subset: probe of 2-1.1:1.0 failed with error -22
[  114.567635] cdc_subset 2-1.1:1.1 usb0: register 'cdc_subset' at usb-0000:00:1d.0-1.1, Linux Device, 42:86:c5:36:15:52
[  114.567689] usbcore: registered new interface driver cdc_subset
[  114.569613] cdc_ether: probe of 2-1.1:1.0 failed with error -16
[  114.569650] usbcore: registered new interface driver cdc_ether
```

Opalinus, 2024-01-19:
> > Your network device name should start with "en". Post your dmesg output to debug your problem.  
> 
> ```
> doas dmesg -W
> [   89.299006] logitech-hidpp-device 0003:046D:4096.0009: HID++ 4.5 device connected.
> [  114.430881] usb 2-1.1: new high-speed USB device number 4 using ehci-pci
> [  114.518149] usb 2-1.1: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 5.10
> [  114.518159] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [  114.518163] usb 2-1.1: Product: RNDIS/Ethernet Gadget
> [  114.518166] usb 2-1.1: Manufacturer: Linux 5.10.103-v7+ with 3f980000.usb
> [  114.566557] cdc_subset: probe of 2-1.1:1.0 failed with error -22
> [  114.567635] cdc_subset 2-1.1:1.1 usb0: register 'cdc_subset' at usb-0000:00:1d.0-1.1, Linux Device, 42:86:c5:36:15:52
> [  114.567689] usbcore: registered new interface driver cdc_subset
> [  114.569613] cdc_ether: probe of 2-1.1:1.0 failed with error -16
> [  114.569650] usbcore: registered new interface driver cdc_ether
> ```  
> should note: my host is gentoo linux, kernel 6.5.9, llvm libc

Can you try different usb ports and cables? It should not be failing with errors.

Also too bad we can't see the type of display on the third picture: In the lower left corner there are glue residues but no sticker

formula-spectre, 2024-01-19:
```main.name = "pwnagotchi"
main.whitelist = [
  "redacted"
]

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
  "redacted"
]

ui.display.type = "waveshare_2"
```

I did not know I had V4; I ordered v2... is v4 supported? also yes, I will try with other cables

Opalinus, 2024-01-19:
> ```main.name = "pwnagotchi"
> main.whitelist = [
>   "redacted"
> ]
>
> main.plugins.grid.enabled = true
> main.plugins.grid.report = true
> main.plugins.grid.exclude = [
>   "redacted"
> ]
>
> ui.display.type = [...]

Maybe try with a known good micro usb cable and a micro usb device first. Ensure it is recognized properly by lsusb

It is clearly labeled as V4 of the rev2.1

You are also missing the enable value 🥲

formula-spectre, 2024-01-19:
> Maybe try with a known good micro usb cable and a micro usb device first. Ensure it is recognized properly by lsusb  

the current cable i'm using works fine; I plug my kobo ereader and it functions fine, I can see it in `lsusb`

> It is clearly labeled as V4 of the rev2.1

yeah but the store page said it was v2 lol, do I put `ui.display.type = "waveshare_4"`?

> You are also missing the enable value 🥲  

I did put it in /etc/pwnagotchi/config.toml:
```
# Add your configuration overrides on this file any configuration changes done to default.toml will be lost!
# Example:
ui.display.enabled = true
ui.display.type = "waveshare_2"
main.custom_plugins = "/usr/local/share/pwnagotchi/custom-plugins/"
```
(I mounted the sd card)

formula-spectre, 2024-01-19:
after putting `waveshare_4` in `/etc/pwnagotchi/config.toml` the screen works and I can confirm that I am in MANU mode! tho I still can't ssh into it with password `raspberry`

Opalinus, 2024-01-19:
> after putting `waveshare_4` in `/etc/pwnagotchi/config.toml` the screen works and I can confirm that I am in MANU mode! tho I still can't ssh into it with password `raspberry`

Great that your display works! 

Starting always in AUTO mode can be set with a marker file (don't know thw path by heart).

Ensure to connect the Data cable before plugging in power. After boot you will not be able to connect via usb

formula-spectre, 2024-01-19:
> Ensure to connect the Data cable before plugging in power. After boot you will not be able to connect via usb    

that is what I'm doing; the microusb part is already plugged in rpi, but the USB-A not; I'm plugging in the USB-a, see usb0 appear and try to connect to that;
to be noted:
```
usb0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq state UP group default qlen 1000
    link/ether ee:3a:e3:69:c8:2e brd ff:ff:ff:ff:ff:ff
    altname enp0s29u1u1
    inet 169.254.120.147/16 brd 169.254.255.255 scope global noprefixroute usb0
       valid_lft forever preferred_lft forever
    inet6 fe80::1584:dbeb:2458:edeb/64 scope link 
       valid_lft forever preferred_lft forever
```
this is what you meant when you said it should start with `en`? I try to connect with `ssh ssh pi@169.254.120.147` but nothing..

Opalinus, 2024-01-19:
> > Ensure to connect the Data cable before plugging in power. After boot you will not be able to connect via usb
> 
> that is what I'm doing; the microusb part is already plugged in rpi, but the USB-A not; 

rpi0w2 does not have USB-A connector, only two micro usb. Check [this picture](https://github.com/aluminum-ice/pwnagotchi/assets/1146338/b5a478b1-62d1-4163-b61f-d3ae644da832) to ensure you didn't mix them up. My rpi zero also powers up when connecting DATA. So try it again:
1. unplug both cables
2. only connect data (which must be already plugged in your computer)
3. then connect power cable
4. then ensure your dmesg does recognice cdc_ether connections correctly

> this is what you meant when you said it should start with `en`? I try to connect with `ssh ssh pi@169.254.120.147` but nothing..

No, this is not the correct device. I do not know which service (other than broken windoze services) do give out adresses from 169.254.0.0/16.

You have already shared the failed cdc_ether connection in dmesg. Try fixing that first, before looking further.

formula-spectre, 2024-01-19:
ok so; I tried on another computer, and it did appear as en<something>; i did `ip addr add 10.0.0.1/255.255.255.0 dev en<something` then `ssh pi@10.0.0.2` and it worked! so the problem is not the cable, it's my host; do I need some particular driver? my kernel is not a pre-compiled one and I left out some stuff; what could be the cause?

formula-spectre, 2024-01-19:
post scriptum: can't I make the rpi connect to my wifi so I can ssh in w/o plugging it in my pc?

formula-spectre, 2024-01-19:
if I know what module I need to use to make the rpi behave properly there is no need; i already saw that another pc, on arch linux, it works fine, idk if it's a kernel module i'm missing/not loaded

formula-spectre, 2024-01-19:
`modprobe usbnet` and then `doas ip address add 10.0.0.1/255.255.255.0 dev enp0s29u1u1` managed to make it work! thank you and sorry for the loss of time lmao
i'm super pumped to have my pwnagotchi working now :D

closed 2024-01-19 by formula-spectre