aluminum-ice/pwnagotchi #111: [BUG] can't access bettercap's webui when the pwnagotchi is plugged in PC and it's in MANU mode

formula-spectre, 2024-01-21:
**Describe the bug**
I am able to plug my pwnagotchi (max-pwnroom) to my pc, give it the 10.0.0.1 address and  `ssh pi@10.0.0.2` fine, but I can't reach `http://max-pwnroom.local:8081/` to use bettercap's webui

**To Reproduce**
Steps to reproduce the behavior:
1. plug in the pwnagotchi, wait it for boot and go into MANU
2. `ip addr add 10.0.0.1/255.255.255.0 dev <iface of the pwnagotchi>`
3. ssh into it, check that bettercap is running

**Expected behavior**
I should be able to go to `http://max-pwnroom.local:8081` to access bettercap's webui,but my browser says it can't reach it; `http://10.0.0.2:8081/` presents to me a blank screen..

**Screenshots**
[image]

not a screenshot but to probe that bettercap is running
```
systemctl status bettercap --no-pager -l
● bettercap.service - bettercap api.rest service.
   Loaded: loaded (/etc/systemd/system/bettercap.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2023-05-04 14:17:05 BST; 7min ago
     Docs: https://bettercap.org
 Main PID: 447 (bash)
    Tasks: 10 (limit: 985)
   CGroup: /system.slice/bettercap.service
           ├─447 bash /usr/bin/bettercap-launcher
           └─576 /usr/bin/bettercap -no-colors -caplet pwnagotchi-manual -iface mon0

May 04 14:17:05 max-pwnroom systemd[1]: Started bettercap api.rest service..
May 04 14:17:05 max-pwnroom bettercap-launcher[447]: grep: /sys/class/net/mon0/operstate: No such file or directory
May 04 14:17:05 max-pwnroom rfkill[498]: unblock set for all
May 04 14:17:07 max-pwnroom bettercap-launcher[447]: WARNING: Terminal colors have been disabled, view will be very limited.
May 04 14:17:07 max-pwnroom bettercap-launcher[447]: bettercap v2.32.0 (built for linux arm with go1.21.5) [type 'help' for a list of commands]
May 04 14:17:08 max-pwnroom bettercap-launcher[447]: [14:17:08] [sys.log] [inf] api.rest api server starting on http://0.0.0.0:8081
May 04 14:17:08 max-pwnroom bettercap-launcher[447]: Are you sure you want to quit this session? y/n [14:17:08] [sys.log] [inf] http.server starting on http://0.0.0.0:80
```

**Environment (please complete the following information):**
 - Pwnagotchi version: 1.8.4
 - OS version: gentoo linux
 - Type of hardware: Rasperry Pi Zero 2 W
 - Any additional hardware Used: Waveshare V4 e-ink display

**Additional context**
https://pwnagotchi.ai/usage/#user-interface I tried to follow these steps; I don't know if this forks does things differently, aside from bc webui's port , my browser simply says unable to connect; I can `ping max-pwnroom` fine;
I tried to add  to `/etc/pwnagotchi/config.toml` the following:  
```
ui.web.username = "username"
ui.web.password = "password"
ui.web.enable = true
```
what I did not try was ui.display.type, since I already have the eink one..

effectx-light, 2024-01-23:
Your Log tells you that he open Bettercap on Port 80. Please Check it.

`May 04 14:17:08 max-pwnroom bettercap-launcher[447]: Are you sure you want to quit this session? y/n [14:17:08] [sys.log] [inf] http.server starting on http://0.0.0.0:80`

formula-spectre, 2024-01-24:
I actually haven't tried that, but the fact that I can simply put http://max-pwnroom.local is related? when I get home I will try to `http://10.0.0.2:80`

formula-spectre, 2024-01-24:
okay my bad, I can access the bettercap web ui via `http://10.0.0.2:80`, but still, `http://max-pwnroom.local` does not correspond to that; do I need to add that to `/etc/hosts`?

closed 2024-03-21 by formula-spectre