aluminum-ice/pwnagotchi #114: [BUG] RPI4 Network is down

Dmario-dev, 2024-01-23:
Hi,
I have instaled the newest version or RPi4.

Interface is booting and the message is "AI Ready" but no networks are discoverd.

In the log I get:
error while setting interface mon0 in monitor mode: No such device exists
error while starting signaling: error while activating handle: No Such Device

So I did:
iwlist wlan0 scan
 >> 
wlan0     Failed to read scan data : Network is down

sudo ifdown wlan0
sudo ifup wlan0

After this (once) I got the list of SSID with iwlist wlan0 scan but nothing in the interface.
So I rebooted the device and I get the same thing from the begining.

In Raspi-config >> System Options >> Wireless LAN 
I get:
Could not communicate with wpa_supplicant

I tried another distribution on this machine and WiFi is working.

There is information that support on RPi4 is provided via firmware patch bcm43455c0/7_45_206/ but I assume this is done in the image?
Any idea how to fix it?

Ravioli5, 2024-02-02:
I have the same problem

MARPATdroid, 2024-02-10:
I figured out a work around.  To be fair I am a network engineer by trade, not a Linux guru at all...  But I'm not afraid to break things so I dove down this rabbit hole.   It appears to be an issue with nexmon and the wireless firmware in the delivered image not supporting monitor mode.   
I did some poking around and also discovered that wpasupplicant was messing with bettercap's ability to put the interface into monitor mode, however this could be a red herring and might have been from some of the other steps I took to troubleshoot.   
However here is my write up for how I resolved this issue.   
This is my first real foray into pwnagotchi, and my first ever real git lab deep dive.  I hope I formatted the information in a way that is useful to people.

#Troubleshooting and isolation Commands

#	sudo ip link set wlan0 down
#	sudo iw wlan0 set monitor control
#	sudo ip link set wlan0 up

#	journalctl -fu bettercap

#	wpa_supplicant -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0 -d

#Chipsets needed to be supported
#	bcm43436b0   //PiZ2W
#
#	bcm43455c0  // Pi4

#Steps I took 
	#Installed packages
	#I did it all as root, I know not great practice, add sudo if you want to be all correct and Linux+ on this
	
	```

	apt install raspberrypi-kernel-headers git libgmp3-dev gawk qpdf bison flex make autoconf libtool texinfo lshw hwinfo

	git clone https://github.com/seemoo-lab/nexmon.git

	cd nexmon

	dpkg --add-architecture armhf

	apt-get update

	apt-get install libc6:armhf libisl23:armhf libmpfr6:armhf libmpc3:armhf libstdc++6:armhf   
	# Failed on libisl23:armhf 
	# No effect on outcome

	ln -s /usr/lib/arm-linux-gnueabihf/libisl.so.23.0.0  /usr/lib/arm-linux-gnueabihf/libisl.so.10
	# ln: failed to create symbolic link '/usr/lib/arm-linux-gnueabihf/libisl.so.10': File exists 
	# No effect on outcome

	ln -s /usr/lib/arm-linux-gnueabihf/libmpfr.so.6.1.0 /usr/lib/arm-linux-gnueabihf/libmpfr.so.4
	# ln: failed to create symbolic link '/usr/lib/arm-linux-gnueabihf/libmpfr.so.4': File exists
	# No effect on outcome

	source setup_env.sh

	make

	cd /root/nexmon/patches/bcm43455c0/7_45_206/nexmon

	make

	make backup-firmware

	make install-firmware

	cd ~/nexmon/utilities/nexutil/

	make && make install

	apt-get remove wpasupplicant

	cd /lib/modules/5.10.103-v7l+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/

	mv brcmfmac.ko brcmfmac.ko.orig

	cp /root/nexmon/patches/driver/brcmfmac_5.10.y-nexmon/brcmfmac.ko .
	
	reboot

```

#Sources:
#	Probably all you need as a solution -> https://github.com/seemoo-lab/nexmon
#
#	Manual monitor mode Pi4 -> https://forums.raspberrypi.com/viewtopic.php?t=253695
#	
#	Obscure line that told me everytime the wireless firmware updated nexmon needed to be re-added -> https://forums.raspberrypi.com/viewtopic.php?t=253695

Ravioli5, 2024-02-12:
Ok I will try