aluminum-ice/pwnagotchi #115: is the UPS lite V1.2 supported? how much does it change from the 1.1?

formula-spectre, 2024-01-25:
I bought an ups lite v1.2, but the ups plugin does not work; if I enable it my pwnagotchi gets stuck in auto mode and the (e-ink) screen is half drawn, and it's constantly sleeping; could it be a faulty pogo pin or is the V1.2 not implemented yet?

formula-spectre, 2024-01-25:
```
main.plugins.ups_lite.enabled = true
main.plugins.ups_lite.shutdown = 2
```
setting the first one to true results into a pwnagotchi like this; I can't even go in MANU mode, I have to insert the sd card, mount rootfs and then put `ups_lite.enabled = false` manually  

[image]

shayna-ilya, 2024-02-10:
I have the same problem. Does anyone know the solution?

formula-spectre, 2024-02-11:
tried in the latest release (1.8.5), still persist. at this point I think it's either software or the UPS lite that i ordered is funny lol

formula-spectre, 2024-02-24:
okay, so, I tried to redo the soldering joints, cleaned them all good, cleaned the pogo pins, I did everything (with the help of an expert in soldering), but nothing. I think the 1.2 changes so much that the evilsocket's plugin for the ups_lite is not capable for it.  
what do we do now? I just wanted a battery percentage level man ;-;  
tl;dr nope, it does not work after redoing the solder joints and making sure the pogo pins make good contact

formula-spectre, 2024-02-24:
p.s. is there any other way I can display the battery percentage? I am interested in only that, like the piSugar's plugin

formula-spectre, 2024-03-30:
folks, I came to an end.
check the back. if it says made by "MJ" get a refund, and buy the original from tindie.
also, use the upslite-v1.3 plugin; the ups1.2 has been obsoleted by the creator
[link to the real UPS Lite, made by the original creator](https://www.tindie.com/products/rachel/ups-lite-for-raspberry-pi-zero/)

closed 2024-03-30 by formula-spectre