aluminum-ice/pwnagotchi #118: SSH credentials (default.toml)

zloneotm, 2024-02-10:
Raspberry Pi 3 here. I'm hoping someone can help me, it must be more simple than I'm thinking. 
I found the info to log in via ssh 
(Changeme) 
I didn't edit the default.toml 
Only created a config. 
As I searched the code the only login I found was changeme 
I'm able to ping my pwnagotchi and bring up a terminal in puTTy but not log in :/ 
I need to ssh in :(
Everything else works though, right after flash 
Thank y'all!!!

zloneotm, 2024-02-10:
Tried pi + raspberry, nothing

zloneotm, 2024-02-11:
> https://pwnagotchi.ai/configuration/#connect-to-your-pwnagotchi

I get your frustration with a simple issue I'm having, but I did do research, believe me. All available tips/tutorials are with the pi 0w. This issue is in regards to my Pi 3B. I'm able to SSH through PuTTy and get to the login, however my issue is with the credentials. Unless a simple data cable is a fix which I highly doubt as I should be able to use the Ethernet. Also had a question about drivers. 
The RNDIS driver I believe is not applicable to the Pi3 
So thank you for trying to redirect me to the simple guide.

MARPATdroid, 2024-02-12:
> I get your frustration with a simple issue I'm having, but I did do research, believe me. All available tips/tutorials are with the pi 0w. This issue is in regards to my Pi 3B. I'm able to SSH [...]

I'll try my hand at helping...  I just got this up and running on a Pi4 so maybe I can help... when you SSH via Putty do you get prompted for a username and password?   The default logins would in fact be "pi" "raspberry" for the initial log in.   

My question for the Pi3 would be are you sure that you have the correct IP that you are trying to get into?    Did you check your DHCP server to see what it is assigning?

MARPATdroid, 2024-02-12:
> I'll try my hand at helping... I just got this up and running on a Pi4 so maybe I can help... when you SSH via Putty do you get prompted for a username and password? The default logins would in fact [...]

Just verified on Pi3B flashed fresh image to 16G SD card.
Put SD card into Pi3B
Connect Cat5 to devnet
Plugin Power
Wait 2 minutes
Watch DHCP requests
Launch Putty
SSH to address
user Pi
Pass Raspberry
Profit.

Close ticket.

zloneotm, 2024-02-12:
ssh is connected just not able to login with pi + raspberry 
will try to add user.conf file with login to see if it helps my case
This issue has only occurred to me with the pi3 (used from ebay)
not pi 0
other than ssh login, ai is working beautifully 
thanks though!

closed 2024-02-14 by zloneotm