aluminum-ice/pwnagotchi #120: Support USB Wifi Dongles (Realtek)

widhalmt, 2024-02-16:
**Is your feature request related to a problem? Please describe.**
I bought a TP-Link Archer T3U Plus because it *should* be supported by kernel 5.10. I even see that it's recognized when I connect it to the pwnagotchi. But Raspian doesn't create the interfaces.

```
Feb 16 11:09:18 *** kernel: [ 8096.338523] usb 1-1.1.2: New USB device found, idVendor=2357, idProduct=0138, bcdDevice= 2.10
Feb 16 11:09:18 *** kernel: [ 8096.338552] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb 16 11:09:18 *** kernel: [ 8096.338566] usb 1-1.1.2: Product: 802.11ac NIC
Feb 16 11:09:18 *** kernel: [ 8096.338579] usb 1-1.1.2: Manufacturer: Realtek
Feb 16 11:09:18 *** kernel: [ 8096.338592] usb 1-1.1.2: SerialNumber: 123456
```

But no new interface is being activated.

**Describe the solution you'd like**
I want to use this external dongle for monitoring with bettercap and pwnagotchi. And have the devices activated automatically when I connect the dongle.

**Describe alternatives you've considered**
I tried to disable onboard wifi with `/boot/config.txt`. This works but it results in no wifi interface being active at all. I'm totally ok if pwnagotichi will only accept one interface. (onboard or, if onboard is deactivated, usb)

**Additional context**

If you need more information, please let me know.

formula-spectre, 2024-02-19:
hey quick question: are you using an usb-A to micro usb adapter for this antenna? sorry if it has nothing to do with the issue, just curious, because I too wanted to put an antenna on my pwnagotchi but did not want to perma-solder it

widhalmt, 2024-02-22:
No, sorry, I should have said that: I'm running on a Raspberry Pi 3. So I can connect it directlly.

sylvainsf, 2024-03-13:
@formula-spectre I'm using a Geekworm Raspberry Pi Zero 2W UPS Expansion Board X306 & USB HUB which is a no solder no header expansion with an 18650 slot and adds 3 USB A ports, might be worth considering.