aluminum-ice/pwnagotchi #122: does/will this project support the orange pi zero 2W?

formula-spectre, 2024-02-19:
https://www.geeky-gadgets.com/orange-pi-zero-2w-features/  
i think this might be a big step forward for the pwnagotchi scene, since it's somewhat more powerful than an rpi02W and it has usb type C.  
I understand if the project will stick to rpi0, but it's neat to consider.  
I think this project _might already_ support this sbc, since at the end, it's a rpi0 clone, but since I don't have one,I can't confirm.  
what do you think?

NMarkgraf, 2024-02-21:
Invest the approx. 20 bucks, test it and report. That would be an idea, wouldn't it? ;-)

formula-spectre, 2024-02-21:
already ordered one ;) tho idk when it will arrive; do I close this issue then re-open it when I get it?  
_insert cave jhonson's quote about science here_

formula-spectre, 2024-04-20:
okay so I tried to flash an image, and in `/boot/config.toml` i've put this basic configuration:
```
main.name = "ORANGE ARMS!"
main.lang = "en"

main.plugins.grid.enabled = true
main.plugins.grid.report = true
main.plugins.grid.exclude = [
mynetworks
]

ui.web.enabled = true
ui.web.address = "::"
ui.web.username = "Gaim"
ui.web.password = "lockon!"
ui.web.origin = ""
ui.web.port = 8080
ui.web.on_frame = ""

ui.display.enabled = false
```

but after a good 20 minutes nothing appeared in my dmesg, aside from the usb device being plugged, and in `ip a` no usb0; am I missing some crucial step?